google/brax #3: Changing `const system` declaration in `html.py` to `var system`?

danielsuo, 2021-06-24:
Hello from the Princeton office! We are excited about `brax` and have started playing around with it.

One snag in our workflow: the generated JavaScript for `html.py:render` declares `system` as a `const`, so re-evaluating an Jupyter notebook cell calling `html.render` will fail. We've forked and changed the `const` to `var`, but hesitated submitting a PR since there may have been some important reason we missed for choosing `const`.

Anyway, just wanted to check!

erikfrey, 2021-06-24:
Hello Daniel - I've been meaning to chase down why rendering fails from time to time.  Thank you for finding that!  A PR would be greatly appreciated.

As for the reason, it's const, programmer error      :-)

danielsuo, 2021-06-24:
Done!

We were also wondering if you'd be amenable to (at least occupying) brax on [pypi](https://pypi.org/project/brax). Happy to set that up and pass ownership to someone on your team if that helps!

(But of course, there are also reasons not to do this, so defer to you guys! Our only reason is to make our `setup.cfg` minimally more clean :) )

danielsuo, 2021-06-24:
Oops that probably should go in another issue...

erikfrey, 2021-06-24:
That's a very kind offer.  Yes please do open another issue and if you add a bit of info about how you did it for deluca, we'll happily follow suit.

PR #4 (2021-06-24): Change `const system` declaration to `var system`
Fixes #3
closed 2021-06-24 by erikfrey