google/brax #5: Add PyPI package for `pip install`

danielsuo, 2021-06-24:


danielsuo, 2021-06-24:
We currently do this manually for `deluca`, but set up the automated process using a `.github/workflow` to upload each time we push a tagged version to GitHub.

I can do it manually for now and we can get more sophisticated as time goes on?

danielsuo, 2021-06-24:
Here's an example: https://pypi.org/project/brax/

Let me know if there's a PyPI user for me to add ownership for!

erikfrey, 2021-06-24:
Thanks so much for doing this!  This will make installing brax easier for folks.

Please add ownership for PyPI user 'erikfrey'

danielsuo, 2021-06-24:
Done!

erikfrey, 2021-06-24:
Thanks again!

erikfrey, 2021-06-24:
Hi @danielsuo - I noticed the pypi release has the directory `brax/envs` but no others.  It should have:

```
brax/envs
brax/training
brax/io
brax/tools
brax/tests
brax/physics
```

Do you know why that is?  Let me know how I can help.

danielsuo, 2021-06-25:
Ah! That's because none of these directories have an `__init__.py` file (even a blank one). It is totally my fault for not raising it earlier. I thought it might be intentional, but of course, the package would break without at least `physics`.

Simply adding an empty `__init__.py` file would resolve these issues. Shall I add?

(Of course, this would also bump us to 0.0.3 at least...FWIW, many Python projects I've seen expend their early patch numbers to PyPI uploading issues...perhaps I am the common factor...)

erikfrey, 2021-06-25:
Derp, good catch.  Happy to add the __init__'s on Tuesday, but if you're blocked feel free to open the PR now and I'll review today, thanks.  No big deal bumping patch numbers - we could bump them to infinity and it won't bother anyone if it's not a breaking change.

danielsuo, 2021-06-29:
Put up 0.0.3!

erikfrey, 2021-06-29:
Just tried it out and works great - did `pip install brax`, then imported it, created an env, and stepped through it.  Thanks!

closed 2021-06-29 by erikfrey