google/brax #8: Effect of batch_shape argument in Envs.create_env

lweitkamp, 2021-07-04:
Hello, working with version 0.0.3. I have a question that concerns the `batch_shape` argument; basically how does it work exactly?

Regardless of the batch_size arg, it seems like it just looks at the first dim of the PRNGKey. For example, with a single key it gives a state batch shape of 2:

```
import jax
import brax

from brax import envs

rng = jax.random.PRNGKey(0)

env = envs.create(env_name='halfcheetah', batch_size=100)
state = env.reset(rng=rng)

print(rng.shape)
print(state.obs.shape)

> (2,
> (2, 23)
```

If i instead split a key in 50 i get the following:

```
import jax
import brax

from brax import envs

rng = jax.random.split(jax.random.PRNGKey(0), 50)

env = envs.create(env_name='halfcheetah', batch_size=100)
state = env.reset(rng=rng)

print(rng.shape)
print(state.obs.shape)

> (50,)
> (50, 23)
```

In both cases the batch_shape argument had no effect. How should I properly create multiple environments on a single device?

cdfreeman-google, 2021-07-04:
Ah, this appears to be something of a sharp edge with our api at the moment.  To answer your immediate question, if you feed in `rng` with the "expected" shape (i.e., 100), everything works fine:

```
import jax
import brax

from brax import envs

batch_size = 100

rng = jax.random.split(jax.random.PRNGKey(0), batch_size)

env = envs.create(env_name='halfcheetah', batch_size=batch_size)
state = env.reset(rng=rng)

print(rng.shape)
print(state.obs.shape)

>>(100, 2)
>>(100, 23)
```

What's going on, here, is that upon initialization, we determine whether or not to create vectorized versions of the reset and step functions based on whether a batch size has been set (snippet below, [this line](https://github.com/google/brax/blob/main/brax/envs/env.py#L54): 

```
  def __init__(self,
               config: brax.Config,
               batch_size: int = 0,
               action_repeat: int = 1,
               episode_length: int = 1000):
    config.dt *= action_repeat
    config.substeps *= action_repeat
    self.sys = brax.System(config)
    self.batch_size = batch_size
    self.action_repeat = action_repeat
    self.episode_length = episode_length

    if batch_size:
      self.reset = jax.vmap(self.reset)
      self.step = jax.vmap(self.step)
```

Note that in JAX, the _actual_ batch dimension of these functions is determined at evaluation time--`jax.vmap`-ing the function, here just adds a leading batch dimension to the evaluation signature of the function, it doesn't actually say what that batch dimension _should_ be.  Thus, when you call these functions with `rng` that has a different shape, it simply returns `state` with same shape semantics as the args you fed it.  Thus, the leading dimension shape of `rng` sets the size of the leading dimension returned.

Let me noodle for a bit on the best way to handle this.  This isn't exactly "unexpected", because these `vmap`'d functions are behaving how they "should", but it's maybe a little unnerving that `reset` is allowed to return something with a shape different than what the env thinks it should be.  Call this a collision between the pure, stateless paradigm of JAX vs the stateful paradigm of most RL env management 😄

lweitkamp, 2021-07-05:
Right, that does make sense. I guess adding a PRNGKey as private variable only for the reset is not a great solution. By the way, I really love this project so far!

erikfrey, 2021-07-09:
Thanks for pointing this out.  One option would be to deprecate the `batch_size` argument in the env init and introduce something simpler, such as `batched`:

```
env = envs.create(env_name='half_cheetah', batched=True)
```

This might more clearly communicate that the user can actually supply any batch dimension they wish to reset() and step(), and everything will work barring extra jit times if the user keeps passing in different shapes.

erikfrey, 2021-11-11:
With the new environment wrappers in `brax/env/wrappers.py` this issue has been addressed.  You now just pass a single `rng` (without splitting) into `env.reset` regardless of whether or not the env is batched.

Closing this as I think it's addressed, but please do re-open if you have any remaining questions.

closed 2021-11-11 by erikfrey