google/brax #9: Differentiating n-link arm produces NaNs

bayerj, 2021-07-07:
I took the double pendulum from the basic tutorial as a starting point for an n-link, planar robot arm. I made the amount of links customisable and added torque-based actuators to the links.

This worked well for 2  and 3 links, but if I go to 4, calculating the gradient wrt the actuation nans.

[Here is a colab notebook](https://colab.research.google.com/drive/1UImoq0j48u1yARFXk6piXIh-4lwwle8x?usp=sharing) demonstrating the problem. The code can also be found below for completeness. At the very bottom there is also the config.

At the moment, I am not sure if it is my limited understanding of dynamics or if there is a bug in brax, jax or even xla. Someone looking at it would greatly help.

On a side note, the compilation times also appear quite long. I have tested the case on CPU only so far, though.

```python
import functools

import brax
import jax
import jax.numpy as jnp

def make_nlink_arm_system(n_links=2):
    arm = brax.Config(dt=0.01, substeps=5)

    body_names = [
        "anchor",
        *[f"middle-{i}" for i in range(n_links - 1)],
        "end",
    ]

    bodies = []
    for body_name in body_names:
        body = arm.bodies.add(name=body_name, mass=1.0)
        bodies.append(body)
        body.inertia.x, body.inertia.y, body.inertia.z = 1, 1, 1
    bodies[0].frozen.all = True

    for body in bodies[1:-1]:
        cap = body.colliders.add().capsule
        cap.radius, cap.length = 0.5, 1

    for i, (parent, child) in enumerate(zip(body_names[:-1], body_names[1:])):
        joint = arm.joints.add(
            name=f"joint-{i}",
            parent=parent,
            child=child,
            stiffness=10000,
            angular_damping=10,
        )
        joint.angle_limit.add(min=-180, max=180)
        joint.child_offset.z = 1.5
        joint.rotation.z = 90

        arm.joints.append(joint)

        arm.actuators.add(
            name=f"actuator-{i}", joint=f"joint-{i}", strength=100
        ).torque.SetInParent()

    return brax.System(arm)

@functools.partial(jax.jit, static_argnums=(0,))
def perform_rollout(f_step, initial_state, plan):
    def f(state, control):
        next_state = f_step(state, control)[0]
        return next_state, next_state

    _, states = jax.lax.scan(f, initial_state, plan)

    return states

def nans(sys, n_steps):
    initial_state = sys.default_qp()
    plan = jnp.ones((n_steps, 2))
    df = jax.grad(lambda *args: perform_rollout(*args).pos.sum(), argnums=2)

    return df(sys.step, initial_state, plan)

if __name__ == "__main__":
    N_STEPS = 16
    sys = make_nlink_arm_system(4)

    res = nans(sys, N_STEPS)   # contains only nans for me

```

```
bodies {
  name: "anchor"
  inertia {
    x: 1.0
    y: 1.0
    z: 1.0
  }
  mass: 1.0
  frozen {
    position {
      x: 1.0
      y: 1.0
      z: 1.0
    }
    rotation {
      x: 1.0
      y: 1.0
      z: 1.0
    }
    all: true
  }
}
bodies {
  name: "middle-0"
  colliders {
    capsule {
      radius: 0.5
      length: 1.0
    }
  }
  inertia {
    x: 1.0
    y: 1.0
    z: 1.0
  }
  mass: 1.0
  frozen {
    position {
    }
    rotation {
    }
  }
}
bodies {
  name: "middle-1"
  colliders {
    capsule {
      radius: 0.5
      length: 1.0
    }
  }
  inertia {
    x: 1.0
    y: 1.0
    z: 1.0
  }
  mass: 1.0
  frozen {
    position {
    }
    rotation {
    }
  }
}
bodies {
  name: "middle-2"
  colliders {
    capsule {
      radius: 0.5
      length: 1.0
    }
  }
  inertia {
    x: 1.0
    y: 1.0
    z: 1.0
  }
  mass: 1.0
  frozen {
    position {
    }
    rotation {
    }
  }
}
bodies {
  name: "end"
  inertia {
    x: 1.0
    y: 1.0
    z: 1.0
  }
  mass: 1.0
  frozen {
    position {
    }
    rotation {
    }
  }
}
joints {
  name: "joint-0"
  stiffness: 10000.0
  parent: "anchor"
  child: "middle-0"
  child_offset {
    z: 1.5
  }
  rotation {
    z: 90.0
  }
  angular_damping: 10.0
  angle_limit {
    min: -180.0
    max: 180.0
  }
}
joints {
  name: "joint-0"
  stiffness: 10000.0
  parent: "anchor"
  child: "middle-0"
  child_offset {
    z: 1.5
  }
  rotation {
    z: 90.0
  }
  angular_damping: 10.0
  angle_limit {
    min: -180.0
    max: 180.0
  }
}
joints {
  name: "joint-1"
  stiffness: 10000.0
  parent: "middle-0"
  child: "middle-1"
  child_offset {
    z: 1.5
  }
  rotation {
    z: 90.0
  }
  angular_damping: 10.0
  angle_limit {
    min: -180.0
    max: 180.0
  }
}
joints {
  name: "joint-1"
  stiffness: 10000.0
  parent: "middle-0"
  child: "middle-1"
  child_offset {
    z: 1.5
  }
  rotation {
    z: 90.0
  }
  angular_damping: 10.0
  angle_limit {
    min: -180.0
    max: 180.0
  }
}
joints {
  name: "joint-2"
  stiffness: 10000.0
  parent: "middle-1"
  child: "middle-2"
  child_offset {
    z: 1.5
  }
  rotation {
    z: 90.0
  }
  angular_damping: 10.0
  angle_limit {
    min: -180.0
    max: 180.0
  }
}
joints {
  name: "joint-2"
  stiffness: 10000.0
  parent: "middle-1"
  child: "middle-2"
  child_offset {
    z: 1.5
  }
  rotation {
    z: 90.0
  }
  angular_damping: 10.0
  angle_limit {
    min: -180.0
    max: 180.0
  }
}
joints {
  name: "joint-3"
  stiffness: 10000.0
  parent: "middle-2"
  child: "end"
  child_offset {
    z: 1.5
  }
  rotation {
    z: 90.0
  }
  angular_damping: 10.0
  angle_limit {
    min: -180.0
    max: 180.0
  }
}
joints {
  name: "joint-3"
  stiffness: 10000.0
  parent: "middle-2"
  child: "end"
  child_offset {
    z: 1.5
  }
  rotation {
    z: 90.0
  }
  angular_damping: 10.0
  angle_limit {
    min: -180.0
    max: 180.0
  }
}
actuators {
  name: "actuator-0"
  joint: "joint-0"
  strength: 100.0
  torque {
  }
}
actuators {
  name: "actuator-1"
  joint: "joint-1"
  strength: 100.0
  torque {
  }
}
actuators {
  name: "actuator-2"
  joint: "joint-2"
  strength: 100.0
  torque {
  }
}
actuators {
  name: "actuator-3"
  joint: "joint-3"
  strength: 100.0
  torque {
  }
}
dt: 0.009999999776482582
substeps: 5
frozen {
}
```

erikfrey, 2021-07-07:
Hey Justin, thanks for providing a lot of details and an example colab.  Really appreciate it.  Here's what I have for you in return: I noticed three issues with your colab.

1) You are adding joints twice to the system config.  You can remove the `arm.joints.append(joint)` line.  This really points out that we need to improve our config validation code - if multiple entries exist in the config with the same name, bad things will happen.

2) 2 => 4 in `plan = jnp.ones((n_steps, 2))`

3) I disabled self-collision of the system ( `arm.collide_include.add()` at the bottom of `make_nlink_arm_system` ).  This is clearly a short term fix / hack, as it points to something specifically in the capsule <-> capsule collision code that isn't differentiable.

Doing these three things removes NaNs.  So, to summarize, from this issue we we will:

1) Improve config validation to disallow repeating the same name
2) Fix capsule <-> capsule so that it's properly differentiable (the other colliders, such as capsule <-> plane already are)

Thanks so much for this report.  Super helpful!

erikfrey, 2021-07-07:
Oh yes, and we hear you that the jit times are slow.  We're actively working on that with the jax team - will hopefully have some updates soon.

bayerj, 2021-07-07:
Thanks very much for the answer @erikfrey, they make it work! They are great to let me play more. ;)

Looking forward to the changes. I am really excited by this project. Thanks for your efforts. 🙏

closed 2021-07-07 by bayerj

erikfrey, 2021-07-09:
Just FYI - we have improved config validation as of cf9cdce610ffe45ef4c4c4a739200d017760318f and you will now get a helpful exception if you pass in a config with duplicate names.

bayerj, 2021-07-09:
Thanks for the heads up! That is quite useful to catch errors as mine, I suppose.