google/brax #11: Performance nitpick

EelcoHoogendoorn, 2021-07-09:
https://github.com/google/brax/blob/8e58feb923ce86b7b8c7036a05429793bbc3fa65/brax/physics/math.py#L278

Little nitpick but 	
```
S = jnp.array([1., -1., -1., -1.])
def inv_quat(q):
	return q * S
```
Benchmarks as 10% faster on my laptop cpu at least; and I suspect the same would be more true of architectures more aggressively tuned for vectorization. Dont have any experience with TPUs and their compilers, but this formulation would also make it easier for a GPU compiler to get to the GPU-optimal compiled code I imagine.

EelcoHoogendoorn, 2021-07-09:
Interestingly, the GPU execution time seems identical, as tested in a [colab](https://colab.research.google.com/drive/1dAtNKkWONhElZZQ6USEBFTVafEHNdZ8q?usp=sharing), and TPU execuation perhaps marginally faster with the suggested implementation. But im probably going something stupid; even though im testing with a device array and making sure to keep the compilation out of the benchmarking loop, the TPU is way slower than the GPU, which does not seem quite right. Or perhaps its just that the invocation overhead for the TPU is that much higher and I should be pushing the benchmarking loop into JAX as well; but srlsy im inv_quatting millions of quats here...

Oh wait even though its telling me a have a TPU it seems like it sometimes gives me a machine without TPU but running CPU emulation or the like, since the runtimes I was getting were suspiciously similar to the CPU ones. After a reconnect I now get substantially faster runtimes; and about a 20% faster performance using my suggested implementation; on a machine that I suppose actually has a non-emulated TPU. But dont quote me on the repeatability of that... still interesting that the GPU implementation seems not to care; that must mean the compiler must realise it ought to fuse the stacking and the multiplying to avoid unnecessary trips to memory; but if it can do that for the GPU why not the other devices? And why wouldnt you see something closer to a 2x speed difference, when this is a 100% memory bound operation on any of these devices? Oh well...

erikfrey, 2021-07-09:
Hello Eelco - thanks for raising this issue!  I went ahead and incorporated your suggestion into the code, which is now live in this repo.

I tried your improvements on some internal Brax benchmarks and I did not see any changes to our environment steps/sec on TPU - it's possible the perf improvement is overshadowed by other perf bottlenecks.  But in any case, it's cleaner and easier to parse, so the change is well worth it.

Thanks again for the suggestion.

closed 2021-07-09 by erikfrey

EelcoHoogendoorn, 2021-07-10:
Cool; yeah wasnt expecting miracles on a non-synthetic benchmark; just exploring the codebase and got curious since I had a 'thats not how id do it in numpy' moment. But interesting to see that JAX/LAX mostly makes this a non-issue anyway. Definitely keen to develop my sense of how to best approach array-programming in JAX. Very pleasantly surprised as a matter of fact how well it works on accelerators; when I first got into JAX I figured that without large mm/convs or hand-fused kernels it probably wasnt worth bothering with accelerators much, but given that the toolchain got this kind of smarts about loop fusion, and the demonstrated gains on this brax-style code, im definitely impressed with the possibilities.

EelcoHoogendoorn, 2021-08-07:
I was browsing the alphafold codebase, and I found this interesting:

https://github.com/deepmind/alphafold/blob/b88f8dacef5d94e4d3d49613d08523feb20caec1/alphafold/model/r3.py#L17

They seem to prefer a struct-of-arrays memory layout, rather than an array-of-structs; this is generally preferred on GPUs with their SIMD architectures; with all threads executing the same instruction at the same time, this gives you the most coalesed memory accesses, which is a big deal. 

That being said this style does not make the code any prettier imo; and in theory you could imagine it should be possible to write everything in array same style, but merely provide compiler hints about the preferred striding of your arrays somehow. Or even automate this; at least on the GPU I think you could formulate a general rule, stating that any axis being vmapped over, should be a candidate for having its arrays allocated with the stride=1 axis, since they will be good candidates for being able to have a good number of threads all executing the same instruction in parallel. Given that JAX compiles for static array shapes, it should also be able to deduce which axes are good candidates for mapping to a threadgrid, and which axes like 3-or-4 dim contractions for instance, are best unrolled inside the kernel instructions.

Anyway, long topic for rambling, but I found it interesting that the alphafold people settled on a different paradigm.

erikfrey, 2021-08-09:
Hi Eelco - I may be missing a subtle point in what you're saying, but: we use jax's wonderful pytree manipulation to convert arrays of structs to structs of arrays.  We either do this using `tree_flatten` in places, or in our helper classes (such as those found in `actuators.py`, `joints.py`, etc) we do this explicitly by building a flax dataclass with fields that are often arrays.

That said, whether XLA lays these data structures out in the most efficient way for access is not something we've looked at too closely.  If you find opportunities for speeding things up, we'd love to hear about it.

EelcoHoogendoorn, 2021-08-09:
Dont think it related to the pytrees mechanism but I might be missing something myself. 

To be more concrete; I thought it was interesting they define they Vec3 like below (whereas these are simply arrays of shape[-1]==3 in brax):

```Vecs = collections.namedtuple('Vecs', ['x', 'y', 'z'])```

That is, the different components are seperate arrays, with no requirement to be anywhere close in memory. Rather, individual vmapped entries will be contiguous in memory; which is likely the ideal layout at least on GPU architectures where you want all threads in a warp to be executing the same instruction at the same time.

And consequently, rather than using builtin matrix-vector ops, they write them out like so:

```
def rots_mul_vecs(m: Rots, v: Vecs) -> Vecs:
  """Apply rotations 'm' to vectors 'v'."""
  return Vecs(m.xx * v.x + m.xy * v.y + m.xz * v.z,
              m.yx * v.x + m.yy * v.y + m.yz * v.z,
              m.zx * v.x + m.zy * v.y + m.zz * v.z)
```

As per the link to the alphafold github above, this is a deliberate choice which they feel leads to better code generation in brax. They dont get quantitative about it in their comments, nor have I profiled it myself; but their comments on the subject make sense (the builtin linear algebra primitives are not optimized for such operation on small axes). It leads to uglier code and im not sure its worth bothering with; but I thought it was interesting at any rate.

erikfrey, 2021-08-09:
Ah, I am following you now.  Yes, we do actually replace some of the linalg code with our own implementations for just the reason you say.  See, for example:

https://github.com/google/brax/blob/main/brax/physics/math.py#L245

We also work closely with the XLA team, so over time we can push these perf issues upstream to them.

EelcoHoogendoorn, 2021-08-09:
Interesting; though in the case of that det operation, if JAX allocates arrays C-style, youd still have three floats contiguous in memory; so on the GPU those memory accesses wouldnt coalesce into a single operation across the warp. I dont know if thats what JAX does though; ideally it would optimize the stride-allocation with all that stuff in mind, and it wouldn't make the last axis contiguous if that wasn't optimal; and we could just write the most high level clean code and trust the compiler to know whats best for us; but I dont have a very clear mental model of how close JAX is to approximating that ideal.

I made a discussion thread [here](https://github.com/google/jax/discussions/7544) in the hope of gaining more insight; but I suppose indeed the XLA team itself is more likely to have insight on these low level questions.

EelcoHoogendoorn, 2021-08-10:
Tried some benchmarking in this [notebook](https://colab.research.google.com/drive/1dAtNKkWONhElZZQ6USEBFTVafEHNdZ8q?usp=sharing); quite some surprises for me.

There are clear timing differences even in this simple 3d matrix-vector product test. Somewhat to my surprise, it matters most on CPU, where the manual mvp beats either einsum or vmapped @-operator by about 2x; not too shabby. GPU seems to care less, but I see about a 20% advantage of the manual mvp there. That surprises me again, since id expect the crux to be in properly coalesced memory access, and Id expect that to be a bigger deal for such a thoroughly bandwidth limited operation. Cant reliably get the TPU to work in my notebook again, and when it does 'work' its horrendously slow. Not sure whats going on there.

Anyway, still a rather synthetic benchmark, but potential 2x performance differences are interesting to keep an eye on I think.

EelcoHoogendoorn, 2021-08-10:
Added another version, replacing named-tuple with unrolled array access.

```
def rots_mul_vecs_arr(m: jnp.array, v: jnp.array) -> Vecs:
  """Apply rotations 'm' to vectors 'v'."""
  return jnp.array([
        m[0, 0] * v[0] + m[0, 1] * v[1] + m[0, 2] * v[2],
        m[1, 0] * v[0] + m[1, 1] * v[1] + m[1, 2] * v[2],
        m[2, 0] * v[0] + m[2, 1] * v[1] + m[2, 2] * v[2]
    ])
```

This is easily the most performant on CPU, but on par with einsum on GPU. Again attempts to try and control memory striding of the input arrays has no effect. TPU still not working.