google/brax #15: URDF

soldierofhell, 2021-07-21:
Hi, any plans for urdf support (converter)? Regards

cdfreeman-google, 2021-07-21:
Hello!  This is actually something I'm putting together right now.  It should land in the next couple of days, with similar feature coverage to our MuJoCo converter.  I'll add a comment here when it's ready for action.

1seck, 2021-07-25:
Will this include obj-support?

cdfreeman-google, 2021-08-13:
Alright, we've just uploaded a VERY rudimentary URDF loader.  Feel free to ping this thread with further feature requests.

We don't yet support objs, but I suspect we'll eventually get around to adding mesh colliders, which will make it a bit easier.

flobotics, 2021-08-16:
@cdfreeman-google i tested the urdf_converter.py and it doesnt work with my urdf, because my config of a joint had another ordering. I put into https://github.com/google/brax/blob/main/brax/tools/urdf.py in the line 127 this code and then it worked with my urdf file.

```
for j in joint:
        if j.tag == "child":
            child = j.get('link')
            print(f"child-name >{child}<")
        if j.tag == "parent":
            parent = j.get('link')
            print(f"parent-name >{parent}<")
      
      #parent = joint[0].get('link')
      #child = joint[1].get('link')
```

hint for beginners:

Inside brax/tools you can run

`python3 urdf_converter.py --help`

And converting a urdf.xml file to brax config use:

`python3 urdf_converter.py --xml_model_path=/home/ros2/Documents/myRobot.urdf.xml --config_path=/home/ros2/Documents/myRobot.txt`

thanks for that tool

flobotics, 2021-08-16:
i found out that it does not add capsule etc. to colliders{} .  So i changed line 171 in https://github.com/google/brax/blob/main/brax/tools/urdf.py to 

```
#colliders = self.links[node].findall('collision')
    colliders = self.links[node]
```

flobotics, 2021-08-16:
i got it, it uses the geometry inside the collision elements, my problem was i only got visual elements in my urdf file.

So to add "something" to see in brax, you need to specify collision with geometry elements inside your urdf.

cdfreeman-google, 2021-08-16:
Ah yes, we don't have a difference between visuals and colliders, where urdf lets you set each separately.  To be safe, I parsed the collider element because that's what matters most for simulation fidelity.

plancherb1, 2021-09-03:
Quick question -- do you have a minimal example for importing a URDF file -- setting an initial state in minimal coordinates (aka I want to set the joint angles, velocities, and optionally joint torques and have it translated to your notation) and then run forward simulation for X timesteps? I think I started to figure it out but have gotten stuck when trying to set the initial position and extract the final position in minimal coordinates (and the default seems to be NaNs). Some code below:
```
# import URDF
URDF_FILE_DATA = files.upload()
brax_urdf_obj = urdf.UrdfConverter(list(URDF_FILE_DATA.values())[0], add_collision_pairs=False)
# set up CONFIG
_CONFIG = brax_urdf_obj.config
_CONFIG.bodies[0].frozen.all = True # fixed base
_CONFIG.gravity.z = -9.8 # set gravity
_CONFIG.collide_include.add() # ignore collisions
# set simulation config params
_CONFIG.dt = 0.1
_CONFIG.substeps: 10
# construct a system
sys = brax.System(_CONFIG)
qp = sys.default_qp() # THIS PRODUCES NaNs
# simulate
qp, _ = sys.step(qp, []) # GET A DIVIDE BY ZERO ERROR
```

erwincoumans, 2021-09-16:
>> is it possible to provide a state/velocity in minimal coordinates (aka joint angles and velocities)?

Yes, you can use default angles in the config file. For example:

```

defaults {
  qps {
    name: "chassis"
    pos {
      x: 0, z: 2.5
    }
  }
  
} 
defaults { 
  angles { 
    name: "RL_lower_leg_2_upper_leg_joint" 
    angle: { x: -35 } 
  } 
  angles { 
    name: "RR_lower_leg_2_upper_leg_joint" 
    angle: { x: -35 } 
  } 
  angles { 
    name: "FL_lower_leg_2_upper_leg_joint" 
    angle: { x: -35 } 
  } 
  angles { 
    name: "FR_lower_leg_2_upper_leg_joint" 
    angle: { x: -35 } 
  } 

  
  }
```

1seck, 2021-09-22:
Is obj import functioning now?

erwincoumans, 2021-11-01:
Since we have basic URDF conversion in Brax, closing for now. If there are specific features missing, please create a new issue (and attach zipped urdf file etc).

closed 2021-11-01 by erwincoumans