google/brax #18: Inverse dynamics?

samuela, 2021-07-26:
Brax (obviously) supports forward dynamics, but I was wondering if it has any support for inverse dynamics as well? I'm looking for something akin to MuJoCo's http://www.mujoco.org/book/APIreference.html#mj_inverse. Looking through the paper and code, I have not been able to find anything yet.

Assuming this is not yet implemented, what do you think it would it entail to implement it?

cdfreeman-google, 2021-08-02:
Hello! We do not currently have this implemented.  Do you have a particular use case in mind?  Implementing inverse dynamics in general is fairy nontrivial, but you might be able to get away with something like a shooting method in the mean time.

samuela, 2021-08-03:
I was hoping to use brax for some kind of motion planning problem, where one might attempt to connect nearby configurations without having a known forward dynamics relationship between them. Ie, it would be very useful for reachability queries.

I'm not familiar with the implementation details of `mj_inverse` but I'm assuming it's likely some kind of shooting method anyhow?

cdfreeman-google, 2021-08-03:
Makes sense.  If it's useful, I hacked up an "inverse dynamics" solution to the reacher task, which should be somewhat adaptable to other tasks (though it's pretty sensitive to how you regularize the shooting).

```import flax
import brax
import jax
from brax import envs

@flax.struct.dataclass
class StepData:
  """Data container for a jax.lax.scan."""
  qp: brax.physics.base.QP
  act: jnp.ndarray

def qp_dist(qp, qp_target, velocity_reg = .01):
  poses = jnp.linalg.norm(qp_target.pos - qp.pos) 
  vels = jnp.linalg.norm(qp_target.vel - qp.vel) 
  angs = jnp.linalg.norm(qp_target.ang - qp.ang)
  rots = jnp.linalg.norm(qp_target.rot - qp.rot)
  return poses + rots + velocity_reg * jnp.linalg.norm(qp.vel)*jnp.linalg.norm(qp.vel) + velocity_reg * jnp.linalg.norm(qp.ang)*jnp.linalg.norm(qp.ang)
  
qp_dist = jax.vmap(qp_dist, in_axes = [0, None])

def inverse_dynamics(step_fn, init_qp, target_qp, max_steps = 100, num_shooters = 1024, seed = 0):

  best_actions = []
  intermediate_qps = []
  min_distances = [10.]

  act_key, next_key = jax.random.split(jax.random.PRNGKey(seed), 2)
  cur_qp = init_qp

  def shoot(vars, index):
    act_key, cur_qp = vars
    act_key, next_key = jax.random.split(act_key, 2)
    cur_act = (jax.random.uniform(act_key,(num_shooters,2))-.5)
    next_qp, _ = jit_env_step(cur_qp, cur_act)
    distances = qp_dist(next_qp, target_qp)
    best_val = jnp.argmin(distances)
    best_dist = distances[best_val]

    this_qp = brax.physics.base.QP(pos = next_qp.pos[best_val],
                            vel = next_qp.vel[best_val],
                            rot = next_qp.rot[best_val],
                            ang = next_qp.ang[best_val])
    cur_qp = brax.physics.base.QP(pos=jnp.stack([this_qp.pos for i in range(num_shooters)]),
                            vel=jnp.stack([this_qp.vel for i in range(num_shooters)]),
                            rot=jnp.stack([this_qp.rot for i in range(num_shooters)]),
                            ang=jnp.stack([this_qp.ang for i in range(num_shooters)]),)
    return (next_key, cur_qp), StepData(qp=this_qp, act=cur_act)

  _, trajectory = jax.lax.scan(shoot, (act_key, cur_qp), jnp.ones(max_steps))

  return trajectory

env_fn = envs.create_fn(env_name="reacher", batch_size=1024)
env = env_fn()
init = env.reset(jnp.array([jax.random.PRNGKey(i) for i in range(1024)]))
jit_sys_step = jax.jit(jax.vmap(env.sys.step))

# note, qps[-1] here is from a training run where I trained a reacher arm, and ran it in inference until it reached some
# trajectory.  if you don't want to retrain a reacher arm, you just have to pick some target qp to reach
out_data = inverse_dynamics(jit_sys_step, init.qp, qps[-1], max_steps=1000)

visualize(env.sys, [brax.physics.base.take(out_data.qp, 999)])```

samuela, 2021-08-06:
Woah, that's awesome! I'll give it a shot!

erikfrey, 2021-11-11:
Closing this as I think it's addressed, but please do re-open if you have any remaining questions.

closed 2021-11-11 by erikfrey