google/brax #21: wire, rope, force-sensor implementations ?

flobotics, 2021-07-27:
Hi,
are there plans to integrate wire or rope as a body ?  Or does it need to be build with many rigid-bodies and joints ?

Is there a way to measure a force onto a rigid-body (force-sensor) from another rigid body ? I mean, e.g. the humanoid-demo has force-sensors at the feets, to measure with how much force-to-the-ground it runs ? Or at the hands, another way to detect falling.

thanks

cdfreeman-google, 2021-08-02:
Hello!

We don't have any immediate plans to integrate wire/rope bodies, but they can, as you mention, be implemented via rigid bodies and joints.

Re measuring forces on rigid-bodies: you can follow a pattern similar to what we used in humanoid.  We pass some of this information around in the `info` dictionary which is populated in the system `info` function on this line:

https://github.com/google/brax/blob/main/brax/physics/system.py#L101

If you wanted to track other impulses, you could put them in this dictionary.

erikfrey, 2021-11-11:
Closing this as I think it's addressed, but please do re-open if you have any remaining questions.

closed 2021-11-11 by erikfrey