google/brax #25: to_tf_model returns: NotEncodableError

flobotics, 2021-08-07:
hi,
when i want to save it to a tf model with:

```
from brax.io import export
export.to_tf_model( "/content/model/", inference_fn, params, state.obs, state.rng)
```

The error then is:

```
NotEncodableError                         Traceback (most recent call last)

<ipython-input-26-74bd7d32a550> in <module>()
      1 get_ipython().system('pip install tensorflow')
      2 from brax.io import export
----> 3 export.to_tf_model( "/content/model/", jit_inference_fn, params, state.obs, state.rng)

14 frames

/usr/local/lib/python3.7/dist-packages/tensorflow/python/saved_model/nested_structure_coder.py in _map_structure(self, pyobj, coders)
     84         return do(pyobj, recursion_fn)
     85     raise NotEncodableError(
---> 86         "No encoder for object [%s] of type [%s]." % (str(pyobj), type(pyobj)))
     87 
     88   def encode_structure(self, nested_structure):

NotEncodableError: No encoder for object [FrozenDict({
    params: {
        hidden_0: {
            bias: TensorSpec(shape=(32,), dtype=tf.float32, name='args_0/1/params/hidden_0/bias'),
            kernel: TensorSpec(shape=(15, 32), dtype=tf.float32, name='args_0/1/params/hidden_0/kernel'),
        },
        hidden_1: {
            bias: TensorSpec(shape=(32,), dtype=tf.float32, name='args_0/1/params/hidden_1/bias'),
            kernel: TensorSpec(shape=(32, 32), dtype=tf.float32, name='args_0/1/params/hidden_1/kernel'),
        },
        hidden_2: {
            bias: TensorSpec(shape=(32,), dtype=tf.float32, name='args_0/1/params/hidden_2/bias'),
            kernel: TensorSpec(shape=(32, 32), dtype=tf.float32, name='args_0/1/params/hidden_2/kernel'),
        },
        hidden_3: {
            bias: TensorSpec(shape=(32,), dtype=tf.float32, name='args_0/1/params/hidden_3/bias'),
            kernel: TensorSpec(shape=(32, 32), dtype=tf.float32, name='args_0/1/params/hidden_3/kernel'),
        },
        hidden_4: {
            bias: TensorSpec(shape=(14,), dtype=tf.float32, name='args_0/1/params/hidden_4/bias'),
            kernel: TensorSpec(shape=(32, 14), dtype=tf.float32, name='args_0/1/params/hidden_4/kernel'),
        },
    },
})] of type [<class 'flax.core.frozen_dict.FrozenDict'>].
```

What could be wrong ?

thanks

erikfrey, 2021-08-09:
Hello!  Can you show me what your inference function looks like?  Or some kind of simplified repro?

flobotics, 2021-08-09:
hi,
i just tested it with your training demo (https://colab.research.google.com/github/google/brax/blob/main/notebooks/training.ipynb)  . Just add a last code-cell and add:

```
from brax.io import export
export.to_tf_model( "/content/model/", inference_fn, params, state.obs, state.rng)
```

then you get the error message

flobotics, 2021-08-24:
tried with new 0.0.4 and original reacher env and training.ipynb, but now it complains about unexpected keyword use_v2 ?  Here is the output

```
WARNING:tensorflow:Mapping types may not work well with tf.nest. Prefer using MutableMapping for <class 'flax.core.frozen_dict.FrozenDict'>

WARNING:tensorflow:Mapping types may not work well with tf.nest. Prefer using MutableMapping for <class 'flax.core.frozen_dict.FrozenDict'>

WARNING:tensorflow:Mapping types may not work well with tf.nest. Prefer using MutableMapping for <class 'flax.core.frozen_dict.FrozenDict'>

WARNING:tensorflow:Mapping types may not work well with tf.nest. Prefer using MutableMapping for <class 'flax.core.frozen_dict.FrozenDict'>

WARNING:tensorflow:Mapping types may not work well with tf.nest. Prefer using MutableMapping for <class 'flax.core.frozen_dict.FrozenDict'>

WARNING:tensorflow:Mapping types may not work well with tf.nest. Prefer using MutableMapping for <class 'flax.core.frozen_dict.FrozenDict'>

WARNING:tensorflow:Mapping types may not work well with tf.nest. Prefer using MutableMapping for <class 'flax.core.frozen_dict.FrozenDict'>

WARNING:tensorflow:Mapping types may not work well with tf.nest. Prefer using MutableMapping for <class 'flax.core.frozen_dict.FrozenDict'>

WARNING:tensorflow:Mapping types may not work well with tf.nest. Prefer using MutableMapping for <class 'flax.core.frozen_dict.FrozenDict'>

WARNING:tensorflow:Mapping types may not work well with tf.nest. Prefer using MutableMapping for <class 'flax.core.frozen_dict.FrozenDict'>

---------------------------------------------------------------------------

TypeError                                 Traceback (most recent call last)

<ipython-input-5-e53e63964dd5> in <module>()
      1 from brax.io import export
----> 2 export.to_tf_model( "/content/model/", inference_fn, params, state.obs, state.rng)

29 frames

/usr/local/lib/python3.7/dist-packages/jax/experimental/jax2tf/jax2tf.py in _dot_general(lhs, rhs, dimension_numbers, precision, preferred_element_type, _in_avals, _out_aval)
   1617         precision_config_proto,
   1618         preferred_element_type=preferred_element_type,
-> 1619         use_v2=True)
   1620     return res
   1621 

TypeError: dot_general() got an unexpected keyword argument 'use_v2'
```

flobotics, 2021-09-11:
with version 0.0.5 is use this statement (only state.rng to rng)

```
from brax.io import export
export.to_tf_model( "/content/model/", inference_fn, params, state.obs, rng)
```

but i also returns 

`TypeError: dot_general() got an unexpected keyword argument 'use_v2'`

erikfrey, 2022-03-21:
OK, this should be fixed now in 7229ee910aaf6a96a30cf55f50fe345feaa31bc0.  Tensorflow does indeed not like FrozenDict, so in this commit we strip them out.

closed 2022-03-21 by erikfrey