google/brax #26: placing camera in simulation world

flobotics, 2021-08-08:
hi,
is there a way to place a camera and then use this camera image inside the step() and _get_obs() methods ?

cdfreeman-google, 2021-08-09:
Hiya!

We don't currently support visual observations, but this is on our todo-list, as it's a fairly common ask.  Will update this issue once in lands, probably within the next month.

flobotics, 2021-08-09:
sounds good

flobotics, 2021-08-16:
if you want to test the trained model in the real world, its very hard to setup the realworld-camera in the same angle/position as the camera in the simulation. 

It would be nice if the user could e.g. define a sphere/box around the simulated-camera-placement and in training the simulated-camera-position changes slightly inside the defined sphere/box. So the model will get more generic.

Perhaps the user need also to define a e.g. arrow that points from the sphere/box to the point-of-interrest and then the camera-position can change but it still knows where to point to the camera.

Another thing with camera too would be to add a plane where pictures can be displayed. These planes could then be used as ground or walls,  so that in every new training episode another picture could be displayed and the camera can do Domain-randomization, also nice for generic models.

mweiss17, 2021-08-31:
Hello,
Just posting to mention that I'm also interested in having visual observations  -- ideally rgbd -- thanks for your work!
Martin

erwincoumans, 2021-11-01:
There is a slow cpu camera implementation that you could try out, based on PyTinyRenderer. This is the same renderer using in PyBullet, when no GPU hardware is available.

See https://github.com/google/brax/blob/main/brax/io/image.py and this [colab](https://colab.research.google.com/github/google/brax/blob/main/notebooks/environments.ipynb), scroll down to the render image function.

a [colab how to use pytinyrenderer ](https://colab.research.google.com/drive/15bxa1v_exEg85_nAwcD46EFF4wqoK25Y?usp=sharing)standalone.

Closing for now. If this doesn't work for you, please re-open.

closed 2021-11-01 by erwincoumans