google/brax #27: Config format, Denavit-Hartenberg, UR5e

bayerj, 2021-08-10:
Hey,

I am currently playing a bit with implementing robot arms that I have given by the Denavit-Hartenberg convention accd to Craig. My plan was to write a function that takes such a sequence of joint specifications and then puts together a config where the corresponding arm is implemented and everything nicely connected by capsules.

I got a little confused along the road, as I started out with the official UR5e DH specification (found on [this page](https://www.universal-robots.com/articles/ur/application-installation/dh-parameters-for-calculations-of-kinematics-and-dynamics/)) and tried to match it to the [config of the UR5e environment](https://github.com/google/brax/blob/main/brax/envs/ur5e.py). Specifically, "joint2" in the UR DH parameters has an "a" value of `-0.425`, while the second joint in the config has a `parent_offset.y = 13.8`. From my view, these values should be the same, up to units. 

Another potential explanation is me not understanding how Brax uses configs to build kinematic chains.

A thing that I struggle with is how a joint's axis of rotation maps to the transformation from the two connected bodies' respective frames. In my experience, these rotations are often given as the angle between the the z axes of the two consecutive frames around the x-axis of the first frame. 

I am sorry that I don't have very clear questions that allow yes/no answers, but at this point I struggle quite a bit with finding how brax relates two links through a joint. I'd very much appreciate a brief explanation of this.

flobotics, 2021-08-11:
hi,
perhaps it would be nice to enable/disable a small plane/grid where the joint is placed. Also nice would be to place at an edge of the small plane/grid three arrows for x,y and z direction.

erikfrey, 2021-11-11:
Agree the joint spec can be confusing!  We're tracking this over at #112 and will be making improvements to the config documentation soon.

closed 2021-11-11 by erikfrey