google/brax #28: support for new environments

optimass, 2021-08-11:
Hi, 

Thanks for the awesome work, I really think this is a game-changer!
Do you have any potential robotics manipulation environments on your roadmap, e.g. [Meta-World](https://meta-world.github.io/)?
I think it would be a great service to the community (including me!) to add one of those.
Maybe it's a daunting task, I can't tell.

Thanks anyways,
Massimo Caccia

erikfrey, 2021-08-12:
Hello Massimo!  We have a few primitive environments that show that brax can manage robotic manipulation, such as [Grasp](https://github.com/google/brax/blob/main/brax/envs/grasp.py).  Our immediate roadmap is to improve some of the core functionality of brax - we have some exciting features coming up - so nothing short term to add new environments.

That said, we don't think building a more robust environment suite would be a daunting task.  Most of the ingredients are there!  So I'm going to leave this issue open in case anyone wants to jump in and propose such a thing.

optimass, 2021-08-12:
Ok great, thanks Erik!

looping @lebrice if you are curious

flobotics, 2021-08-30:
more envs would be great, but would it be not greater to have a tool to build own/custom envs and own/custom robots/models/bodies ?

There are tools like gazebo that has a model-editor ( http://gazebosim.org/tutorials?tut=model_editor ) which saves only sdf format.

Or a solidWorks plugin ( https://blogs.solidworks.com/teacher/wp-content/uploads/sites/3/WPI-Robotics-SolidWorks-to-Gazebo.pdf )  which seemed to only save urdf (but i dont know). Also solidWorks is no free.

Then there are tools to convert e.g. urdf to sdf . E.g. directly from gazebo with gz sdf xx.urdf yy.sdf . But there the problem is, there is no tool to visually create urdf models.

So the question would be, create a tool/plugin to visually create brax-config models? Own tool, or plugin of e.g. freeCAD, skechtup, etc.

 Or create a tool that converts e.g. gazebo sdf files (gazebo has a good model editor, with placing joints/rotation etc.) to brax-config-files like the urdf_converter.py ?

erikfrey, 2021-11-11:
Closing this as I think it's addressed, but please do re-open if you have any remaining questions.

closed 2021-11-11 by erikfrey