google/brax #29: Support for height maps and factor of 2.0 for grip

o-Oscar, 2021-08-12:
Hi,

After close inspection of the collider code, I found this interesting piece of code : 
```python
  # factor of 2.0 here empirically helps object grip
  # TODO: expose friction physics parameters in config
  return dp_n * colliding_n + dp_d * colliding_d * 2.0
```
This factor of two is not physical at all and after some tests, is only non-physics-breaking when gravity is collinear with the collision normal. 

Can someone give some pointers as to why is was put there, home much more "grip" we get with such an hack and whether or not removing it would break performances of the simulator ?

Thanks.

PS : I am implementing, and almost ready to merge-request, support for height-maps : same functionality as planes but with the added benefit of being able to add terrain altitude. Due to the non-vertical normals of such a terrain, this factor of 2.0 comes in conflict with the height map support.

cdfreeman-google, 2021-08-13:
Hello! Adding heightmaps would be extremely cool!  Just saw your PR and it looks great!

This factor came from some old hack-y parity experiments we ran between brax and physX.  Because our contact model is already "unphysical", we weren't too bothered by adding some additional tweaks for getting nicer motion quality.  My plan was to revert this once we did a more careful pass on collision physics in the next few weeks, but if you're already seeing issues, we can definitely revert it earlier (it doesn't really impact perf on learning tasks).

Thanks again for the heightmaps PR!!

flobotics, 2021-08-20:
hi,
how can i use heigthMap in an env ? I tried stuff like

```
bodies {
  name: "ground"
  colliders {
    heightMap {
        size: 10
        data: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
      }
  }
  mass: 1.0
  inertia {
    x: 1.0
    y: 1.0
    z: 1.0
  }
  frozen {
    all: true
  }
}
```

but when i use visualize(env.sys, [state.qp]) to see the heightMap, it only shows a grey image ?  What do i wrong ?

o-Oscar, 2021-08-23:
Yes indeed, the reference to the latest js files still seems to be brocken. For reference, I used the following code in colab for testing :  

```
#@title Colab setup and imports

import jax
import jax.numpy as jnp
from matplotlib.lines import Line2D
from matplotlib.patches import Circle
import matplotlib.pyplot as plt

try:
  import brax
except ImportError:
  from IPython.display import clear_output 
  !pip install git+https://github.com/google/brax.git@main
  clear_output()
  import brax
```
and
```
from IPython.display import HTML, clear_output
from brax.io import html
def visualize(sys, qps):
  """Renders a 3D visualization of the environment."""
  return HTML(html.render(sys, qps))

#@title A bouncy ball scene
bouncy_ball = brax.Config(dt=0.05, substeps=100)

# ground is a frozen (immovable) infinite plane
ground = bouncy_ball.bodies.add(name='ground')
ground.frozen.all = True
plane = ground.colliders.add().heightMap
plane.size=10
plane.data.extend([0]*9)
plane.SetInParent()  # for setting an empty oneof

# ball weighs 1kg, has equal rotational inertia along all axes, is 1m long, and
# has an initial rotation of identity (w=1,x=0,y=0,z=0) quaternion
ball = bouncy_ball.bodies.add(name='ball', mass=1)
ball.inertia.x, ball.inertia.y, ball.inertia.z = 1, 1, 1
cap = ball.colliders.add().capsule
cap.radius, cap.length = 0.5, 1

# gravity is -9.8 m/s^2 in z dimension
bouncy_ball.gravity.z = -9.8

sys = brax.System(bouncy_ball)
qp = sys.default_qp()

visualize(sys, [qp])
```

This does indeed only give me a simple sphere and no height map.

When inspecting the js code loaded onto the colab, we can see that it is an old version of the js, with no support to heightMap.

I will update issue #33 this evening to give a way to use the local version of the js files to allow height map visualization and local js development.

erikfrey, 2021-08-23:
Yes agreed, using the cdn complicates pushing new functionality.  If you update the js location in `html.io` to this:

https://cdn.jsdelivr.net/gh/google/brax@ab65a0356bc837ba32b409410acf534dbe9b115b/js/viewer.js

You will see the new js.

erikfrey, 2021-08-24:
fyi this is updated now - `io.html` uses a newer commit

flobotics, 2021-08-24:
i run this code:

```
#@title Colab setup and imports
#@markdown ## ⚠️ PLEASE NOTE:
#@markdown This colab runs best using a TPU runtime.  From the Colab menu, choose Runtime > Change Runtime Type, then select **'TPU'** in the dropdown.

from datetime import datetime
import functools
import os

from IPython.display import HTML, clear_output
import jax
import jax.numpy as jnp
import matplotlib.pyplot as plt

try:
  import brax
except ImportError:
  !pip install git+https://github.com/google/brax.git@main
  clear_output()
  import brax

from brax import envs
from brax.training import ppo, sac
from brax.io import html

from google.protobuf import text_format
from brax.envs import env
from typing import Tuple

if 'COLAB_TPU_ADDR' in os.environ:
  from jax.tools import colab_tpu
  colab_tpu.setup_tpu()

_SYSTEM_CONFIG = """
bodies {
  name: "Ground"
  colliders {
    heightMap {
        size: 10
        data: [0.0, 0.2, 0.3, 0.5, 0.0, 0.3, 0.0, 0.2, 0.0]
    }
  }
  inertia {
    x: 1.0
    y: 1.0
    z: 1.0
  }
  mass: 1.0
  frozen { all: true }
}

bodies {
  name: "body_0"
  colliders {
    capsule {
      radius: .15
      length: 1.0
    }
    position { 
        x: .00
        z: .0
    }
  }
  inertia {
    x: 1.0
    y: 1.0
    z: 1.0
  }
  mass: 1.0
  frozen { all: true }
}

bodies {
  name: "target"
  colliders {
    position {
    }
    sphere {
      radius: 0.09
    }
  }
  inertia {
    x: 1.0
    y: 1.0
    z: 1.0
  }
  mass: 1.0
  frozen { all: true }
}

joints {
  name: "joint_ground"
  stiffness: 100.0
  parent: "Ground"
  child: "body_0"
  parent_offset {
    z: 0.0
  }
  child_offset {
  }
  rotation {
    y: 90.0
  }

  angle_limit {
      min: -360
      max: 360
  }

  limit_strength: 0.0
}

dt: 0.02
substeps: 4
frozen {
  position {
    z: 1.0
  }
  rotation {
    x: 1.0
    y: 1.0
  }
}
"""

class Skeleton(env.Env):
  """skeleton"""

  def __init__(self, **kwargs):
    config = text_format.Parse(_SYSTEM_CONFIG, brax.Config())
    super().__init__(config, **kwargs)
    #self.target_idx = self.sys.body_idx['target']
    self.target_idx = self.sys.body_idx['body_0']

  def reset(self, rng: jnp.ndarray) -> env.State:
    qp = self.sys.default_qp()
    info = self.sys.info(qp)
    obs = self._get_obs(qp, info)
    reward, done, steps, zero = jnp.zeros(4)
    metrics = {
        'rewardDist': zero,
        'rewardCtrl': zero,
    }
    return env.State(rng, qp, info, obs, reward, done, steps, metrics)

  def step(self, state: env.State, action: jnp.ndarray) -> env.State:
    rng = state.rng
    qp, info = self.sys.step(state.qp, action)
    obs = self._get_obs(qp, info)

    reward = 1.0

    steps = state.steps + self.action_repeat
    done = jnp.where(steps >= self.episode_length, 1.0, 0.0)
    metrics = {
        'rewardDist': 1.0,
        'rewardCtrl': 1.0,
    }

    return env.State(rng, qp, info, obs, reward, done, steps, metrics)

  def _get_obs(self, qp: brax.QP, info: brax.Info) -> jnp.ndarray:
    """Egocentric observation of target and arm body."""
    
    return qp.pos[self.target_idx]

  def _random_target(self, rng: jnp.ndarray) -> Tuple[jnp.ndarray, jnp.ndarray]:
    """Returns a target location in a random circle slightly above xy plane."""
    rng, rng1, rng2 = jax.random.split(rng, 3)
    dist = .2 * jax.random.uniform(rng1)
    ang = jnp.pi * 2. * jax.random.uniform(rng2)
    target_x = dist * jnp.cos(ang)
    target_y = dist * jnp.sin(ang)
    target_z = .01
    target = jnp.array([target_x, target_y, target_z]).transpose()
    return rng, target

config = text_format.Parse(_SYSTEM_CONFIG, brax.Config())

envf = Skeleton()
state = envf.reset(rng=jax.random.PRNGKey(seed=0))

def visualize(sys, qps):
  """Renders a 3D visualization of the environment."""
  return HTML(html.render(sys, qps))

visualize(envf.sys, [state.qp])
```

Then i see that the heigtMap is not in the center, or ? :

[image]

erikfrey, 2022-03-21:
@flobotics we've made some fixes since this issue was opened.  Closing but please let us know if you're still having problems with heightmaps.

closed 2022-03-21 by erikfrey