google/brax #32: Added walls to environment disappear in training

flobotics, 2021-08-15:
hi,
i build a custom env ( https://github.com/flobotics/brax/blob/main/brax/envs/simtoreal.py ) , which is mostly the reacher with walls around it. If i use html.render() it shows everything, but when i then go on with training and then look at the trajectory output, The walls start to sink into the ground and disappear ? I want to use the walls to let the target bounce off the walls , but if they are not there ?  What do i wrong ?

Here a images of the start of the trajectory video output:

[image]

Then it disappears

[image]

till its complety gone

[image]

flobotics, 2021-08-15:
adding frozen { all: true } to the wall bodies and they do not disappear anymore.  But when i now remove frozen { all: true } from the target-sphere-body,  so that the target-sphere-body can move around inside the walls, the target-sphere-body disappears in the trajectory-video-output ?

How can i use a sphere as a ball, jumping around and so ?

flobotics, 2021-08-15:
i found that the ball is falling through the ground/plane at the beginnng of the trajectory video output ? I included 

```
collide_include {
    first: "ground"
    second: "ball"
}
collide_include {
    first: "ball"
    second: "ground"
}
```

But it still falls through ground ? How can i prevent that ?

[image]

erikfrey, 2021-08-16:
Hello!  I think you have found a bug where brax doesn’t properly register the sphere collider.  As a workaround until we put in the proper fix, Can you try turning the ball into a sphere-shaped capsule where the length is double the radius?  See the `Object` body in `grasp.py` for an example.

flobotics, 2021-08-16:
turning the ball from sphere into capsule like you said works, the ball bounces off the ground and rolls to the middle.

Perhaps the next bug is, that it seemed that all bodies or joints depend on ground-joint. I cannot describe it good, but if i lower only the parent_offset z value of the ground-arm_0-joint from 0.1 to 0.01 you see that also the walls and the new capsule-ball get lowered too, and sink into the ground ? But i only wanted to lower-down the arm_0 and arm_1 ?

[image]

erikfrey, 2021-08-16:
cc @cdfreeman-google 

Good timing for pointing this out.  Configs are missing an important piece of information which is to specify an initial offset for an unparented (is not the child of a joint) body.  Right now the system "guesses" where to put these unparented bodies and in your case, the guess is incorrect.

We are adding this new config now and will let you know when it's ready.  After it's done, you'll be able to specify different offsets for your walls vs the ball and arms, to put everything in its right place.

flobotics, 2021-08-16:
i turned the boxes into capsule for the arms, but now the ball hits the ground, then rolls into the middle of the plane and then at the end one arm seemed to kick the ball out of the arena ?

Why are the arms not all the time hit the ball ?

A gif to watch is here: https://github.com/flobotics/brax/blob/main/out.gif

erikfrey, 2022-03-21:
fyi, you can now specify initial positions for bodies in Brax, see for example this test:

https://github.com/google/brax/blob/main/brax/tests/physics_test.py#L128

Closing this issue, but please reach out if more problems remain.

closed 2022-03-21 by erikfrey