google/brax #33: Make it easier to develop local changes to js

erikfrey, 2021-08-19:
`html.io` currently refers to a static cdn for the js, which is awkward for trying out changes to the js itself.

We have a hacky example of local dev of js here: https://colab.sandbox.google.com/gist/erikfrey/eb8b138dce06f7c2135e9d9690a09a8e/local_dev.ipynb#scrollTo=NaJDZqhCLovU

A proper solution might be to introduce an arg to html.render that takes in a different static path, and that starts up a temp webserver.

o-Oscar, 2021-08-23:
For anyone stumbling on this thread, for me the most convenient way to make local dev to js files is to run the following script on brax's root folder : 

```
import http.server

class MyHTTPRequestHandler(http.server.SimpleHTTPRequestHandler):
    def end_headers(self):
        self.send_my_headers()
        http.server.SimpleHTTPRequestHandler.end_headers(self)

    def send_my_headers(self):
        self.send_header("Cache-Control", "no-cache, no-store, must-revalidate")
        self.send_header("Pragma", "no-cache")
        self.send_header("Expires", "0")
        self.send_header("Access-Control-Allow-Origin", "*")

if __name__ == '__main__':
    http.server.test(HandlerClass=MyHTTPRequestHandler)
```

Then is is a simple matter of : 
```
def visualize(sys, qps):
    port = 8000
    to_render = html.render(sys, qps)
    to_render = to_render.replace(
        "https://cdn.jsdelivr.net/gh/google/brax@a93dadc48a41177a4d9e7794ce17c1ad98ed8583", 
        f"http://127.0.0.1:8000"
    )
    display(HTML(to_render))
```

flobotics, 2021-08-30:
hi @o-Oscar ,
when i use this, it does not show anything ?  The webserver is running, i can access the files with a browser.  There are also no error messages ?  When i print(f"html>{to_render}<")  i can see that the string is replaced correctly, but i cannot see anything ?

You got a hint ?

namheegordonkim, 2021-09-13:
FWIW, I've been dumping viewer HTML explicitly using `save_html()` from `html.py`, changing the JS import statement to my local `viewer.js` path, and running viewer.html from a Python simple HTTP server (e.g. `python -m http.server`). I do agree that it's a bit convoluted though. Would be great to have a native viewer window, as brought up in https://github.com/google/brax/issues/47

EelcoHoogendoorn, 2021-10-18:
> hi @o-Oscar , when i use this, it does not show anything ? The webserver is running, i can access the files with a browser. There are also no error messages ? When i print(f"html>{to_render}<") i can see that the string is replaced correctly, but i cannot see anything ?
> 
> You got a hint ?

Ive got the same issue as described here.

EelcoHoogendoorn, 2021-10-21:
Nevermind, worked after more permutations; not sure which ones