google/brax #38: Analytical policy gradient demo

peilundai, 2021-08-25:
Hi, Brax developers, thanks for developing Brax. I saw that analytical policy gradient training function (apg.py) has bee added and updated in the past few days. I have tried to use it to train in some of the given envs (e.g. ant), but it doesn't give me the results expected. I wonder if you could provide a demo setting for using apg in some of the envs.

cdfreeman-google, 2021-08-31:
Hello!  Indeed, we've added a `truncation_length` parameter to ant that makes it a bit easier to optimize.  The following training function will train an ant that can crawl a bit to the right--it's prone to getting stuck once it isn't facing to the right anymore, but it does go significantly farther than the previous no-truncation ant did:

```train_fn = functools.partial(
    apg.train,
    episode_length=400,
    action_repeat=1,
    num_envs=1024,
    num_eval_envs = 128,
    learning_rate=3e-3,
    log_frequency = 300,
    truncation_length = 10,
    normalize_observations = True,
)```

I've had some success by augmenting this with state resetting, so that, say, half of the environments in a batch are reset to wherever the ant goes after one whole episode.  This seems to prevent it from getting stuck as much, but it isn't submitted yet because it's a little bit hacky.

peilundai, 2021-09-03:
Thank you @cdfreeman-google. I tried running using the setting you gave, and with log_frequencey = 300, it takes ~20min, which is much slower than other settings. Is that normal considering Brax is supposed to be very fast on TPU?

cdfreeman-google, 2021-09-03:
Indeed, gradients are a bit slow at the moment compared to the other algorithms.  You can _probably_ get away with training an ant with an episode length of ~100, and maybe ~100 steps (we really should change the log_frequency kwarg to be "training_steps" or something...), which should make things go a bit faster, but for shorter trajectories you almost certainly need more clever reset logic.

peilundai, 2021-09-04:
That makes sense. I am also trying different ways to improve the training performance and speed. Thank you.

closed 2021-09-04 by peilundai