google/brax #44: bug with "mass"

flobotics, 2021-08-29:
hi,
this config

```
_SYSTEM_CONFIG = """
bodies {
  name: "box_1"
  colliders {
    box {
      halfsize {
          x: 0.5
          y: 0.5
          z: 0.5
        }
    }
  }
  inertia {
    x: 1
    y: 1
    z: 1
  }
  mass: 0.03
}

bodies {
  name: "box_2"
  colliders {
    box {
      halfsize {
          x: 0.25
          y: 0.25
          z: 0.25
        }
    }
  }
  inertia {
    x: 1
    y: 1
    z: 1
  }
  mass: 0.03
}
bodies {
  name: "Ground"
  colliders {
    plane {
    }
  }
  frozen {
    all: true
  }
}
bodies {
  name: "target"
  colliders {
    sphere {
      radius: 0.009
    }
  }
  frozen { all: true }
}

joints {
  name: "joint0"
  angle_limit {
      min: -60
      max: 60
  }
  rotation {
    z: 90
  }
  
  parent_offset {
    y: -0.6
  }
  child_offset {
      z: 0.1
  }
  
  parent: "box_1"
  child: "box_2"
  stiffness: 5000.0
  angular_damping: 35
}

actuators {
  name: "joint0"
  joint: "joint0"
  strength: 300.0
  torque {
  }
}

friction: 0.6
gravity {
  z: -9.81
}
baumgarte_erp: 0.1
angular_damping: -0.05

collide_include {
  first: "box_1"
  second: "Ground"
}
collide_include {
  first: "box_2"
  second: "Ground"
}

dt: 0.02
substeps: 4
"""
```

you can see the two boxes with training.ipynb with html.render, but after training if you visualize a trajectory, there is only a grey image to see.

If you stop and reset the trajectory video, you see in the controls the x,y,z, etc. values which are going fastly into minus till NaN, if you play the video?

If you change the "mass" values of both boxes to 1.0 ,  everything looks ok ?

cdfreeman-google, 2021-08-31:
One sharp edge for our engine is that small masses typically need shorter simulation steps.  I'd bet everything would work if you upped the number of substeps (or, equivalently, decreased dt).

flobotics, 2021-09-05:
thanks for help,

it worked , i used dt: 0.015
substeps: 24

and then it shows up

closed 2021-09-05 by flobotics