google/brax #45: endless visualization of trajectory

flobotics, 2021-08-29:
hi,
is it possible to add a feature to display endlessly the trajectory of the learned inference function ? 

At the moment it seemed to be fixed to "20" , i think it is "seconds"  ?

thx

erikfrey, 2021-08-30:
By endlessly do you mean loop?  Our next push of the javascript viewer will have a 'loop' checkbox in the animation control panel.

If you mean actually mean infinite length trajectories, this is on our roadmap but will not happen straight away.  We are refactoring the viewer to support interactive physics, at which point you will be able to stream physics steps/inference indefinitely.

And yes, 20 seconds trajectory is what you would expect if your episode has 1000 frames with .02s dt.

flobotics, 2021-08-30:
i mean infinite length trajectory , if its on the roadmap thats cool.

closed 2021-09-02 by flobotics