google/brax #46: strange physics?

flobotics, 2021-08-30:
hi,

when i use this env https://github.com/flobotics/brax/blob/main/brax/envs/crawler4.py with training.ipynb then i get this strange behaviour . The block is 10kg in weight, the rest only 1.0kg ?  What do i wrong ?

here is a video of it https://github.com/flobotics/brax/blob/main/strange-physics.mp4

cdfreeman-google, 2021-08-31:
You may want to set the `baumgarte_erp` parameter in your config to be .05 or possibly .1.  I _think_ this is because the cube is not be aggressively de-penetrated enough from the ground.  See here for example: https://github.com/google/brax/blob/main/brax/envs/ant.py#L361

flobotics, 2021-08-31:
thanks alot. While experimenting alot i missed baumgarte_erp, friction and gravity in the config. 

But now it only moves the joints into one direction and then it stops doing anything ?  I tested different hyperparams but no changes ?

Here is a video https://github.com/flobotics/brax/blob/main/crawler4.mp4  (click on view raw)

flobotics, 2021-09-02:
i did some new tests with two simple boxes. But i dont get it that it learns to move, it seemed to only rotate into one direction and then stops, or it is dancing ?

I wrote what i have done with some gifs that displays the "progress" :)

You can see it directly online here:
https://github.com/flobotics/custom_ros2_documentation/blob/main/Building-google-brax-environment.md

erwincoumans, 2021-11-03:
Closing since the linked github repo is unavailable. Please re-create a new issue if required, attach zip files instead of web links, so the information keeps available longer.

closed 2021-11-03 by erwincoumans