google/brax #47: Rendering without notebooks

namheegordonkim, 2021-08-31:
Hi all,

Is rendering without notebooks supported? I personally don't like using notebooks and am much more used to working with e.g. GLUT windows. It seems that wrapping brax environment into `gym` as per the third tutorial (https://colab.research.google.com/github/google/brax/blob/main/notebooks/pytorch.ipynb#scrollTo=9mqL2S67losC) doesn't let me call `env.render()` (throws `NotImplementedError`).

Any thoughts would be highly appreciated. Thanks!

kayuksel, 2021-09-01:
Yes, that would be a great feature, even if that means not rendering whole batch but a selected one.
If we could render all of them in a single scene (where they do not collide), that could also be good.

bayerj, 2021-09-07:
I have been using [streamlit](https://streamlit.io/) for showing the renderings. If you have the html, it is as easy as this:

```python
import streamlit.components.v1 as components
html_string = load_html_rendering()    # this would have to be dumped in a different process
components.html(html_string, height=500)
```

Then reloading the streamlit app shows it. That's also nice to show all kinds of other analytics. Here is a screenshot of an app I built:

[image]

erikfrey, 2021-09-11:
We have begun work on this.  No ETA yet but it's an active priority for us.

EelcoHoogendoorn, 2021-09-12:
Interesting; that streamlit thing looks pretty slick.

I was running into the same issues myself; not very familiar with 3JS and I think itd be cleaner to keep the visualization within python. Or at least have multiple rendering options available.

The best thing I can come up with would be to use the high level vispy/scene module; which is a similar high level API to 3JS; though its a bit stale in terms of maintenance... but I suppose its fine for putting some cylinders and boxes on screen. Not sure if there is a better option in the python ecosystem; doing it in raw pyopengl sure would be a kludge.

yardenas, 2022-04-14:
@erikfrey what's the timeline on the GPU rendering project? I might be able to contribute some parts.