google/brax #49: Replacing gym's Mujoco envs with brax envs

vwxyzjn, 2021-09-01:
Had a conversation with @jkterry1 on https://github.com/openai/gym/issues/2366, and it appears brax would also be a great alternative for the mujoco envs replacement. 

To help with this transition. I made an attempt to try out brax with pytorch. Here is a basic report: https://wandb.ai/costa-huang/brax/reports/Brax-as-Pybullet-replacement--Vmlldzo5ODI4MDk. The source code is here: https://github.com/vwxyzjn/cleanrl/blob/mybranch/cleanrl/brax/readme.md

One of the biggest issue with the brax adoption is the env normalization:
* gym doesn't have a normalization wrapper
* sb3 has a normalization wrapper but brax does not have a sb3 vector env api
* brax's normalization is implemented in the training side (https://github.com/google/brax/blob/main/brax/training/normalization.py)

I think going forward, probably the best way to fix this is to refactor the brax training side's normalization to the environment side. This in the future will also help throughput with the `JaxToTorchWrapper`. Otherwise, the observation will go from GPU to CPU for gym or sb3's normalization wrapper, then GPU again for torch, which just doesn't make sense. 

One small thing is that given the brax environment directly produces the vector env, there is also no way to inject a `ClipActionsWrapper(env)`, which may or may not have a performance impact. That said, this can be implemented in the training side with ease.

erwincoumans, 2021-09-01:
Yes, as I suggested previously, Brax seems a good option, since it allows for GPU and TPU accelerators, next to CPU. We can use this issue to track progress and add an itemized todo.

jkterry1, 2021-09-05:
To recap the to do list:

1. Add suitable rendering
2. Further tune observation/action spaces to make them as close as possible
3. Make sure we are not reproducing the list of bugs in MuJuCo environments from Antonin Raffin that I sent you

I feel like there may have been a 4th issue, but I don't sleep very much and can no longer recall it. @erwincoumans @benelot do you remember?

vwxyzjn, 2021-09-05:
One note on the suitable rendering is I feel implementing `env.render(“rgb_array”)` might be too expensive and counterproductive. Maybe implementing `env.render(“html”)` at the end of training is more preferable.

jkterry1, 2021-09-05:
They're planning to add a new rendering engine such that "rgb_array" will be suitable

jkterry1, 2021-09-05:
I don't know if this is the 4th feature I can't remember, but another thing we'll need to deal with that I briefly discussed is action/observation space documentation for the new Gym website we're working on, in the flavor of https://www.pettingzoo.ml/classic/chess

joaogui1, 2021-09-05:
I would like to help with this, what can I do to help?

jkterry1, 2021-09-06:
@joaogui1 Probably nothing, at least at the moment. Right I'm waiting on the Brax team to do some work and for the guy who created the pybullet replacement envs to get back from vacation, this will take 4-6 weeks. If you'd like to help with gym maintenance problems in general though, please email me and we can coordinate some things (jkterry@umd.edu)

joaogui1, 2021-09-07:
Got it, will wait a little then, thanks!

sgillen, 2021-09-08:
I'm also happy to help on this, I've spent a lot of time with the mujoco/pybullet environments at this point. Can certainly help with points 2/3 that @jkterry1 posted in this thread.

erikfrey, 2021-09-11:
We have started working on 1) the renderer.  We're looking at porting a simple technique like https://github.com/rougier/tiny-renderer to JAX as a new module in `brax.io`

Tuning observation/action space could start in parallel if anyone is interested.  I think the steps would involve:

1) reset a Gym Mujoco env (say Ant) to default state and inspect the observation space and its description
2) compare to Brax Ant env and make adjustments
3) step both and compare dynamic observations (e.g. contact forces)

I think the envs are already ~80% comparable, and the last 20% is just sleuthing to read the mujoco docs, and confirm the format matches.  I think we can get to the point where the meaning of each observation dimension is the same in both envs, even if the dynamics are still different.

sgillen, 2021-09-11:
I can get that going next week. I will use Mujoco 1.5 due to [this ]( https://github.com/openai/gym/issues/1541) issue. It looks like the Brax environments are based off the v2 version of the Mujoco environments, so I'll start by comparing to those. Based on https://github.com/openai/gym/pull/1304 I think the v3 versions are supposed to be identical if using default args, not 100% sure that's true though.

vwxyzjn, 2021-09-12:
This is so great to hear! I also have a quick update. Gym now has a normalization wrapper: https://github.com/openai/gym/pull/2387. The usage is roughly
```python
env = gym.make("HalfCheetahBulletEnv-v0")
env = gym.wrappers.RecordEpisodeStatistics(env)
env = gym.wrappers.ClipAction(env)
env = gym.wrappers.NormalizeObservation(env)
env = gym.wrappers.TransformObservation(env, lambda obs: np.clip(obs, -10, 10))
env = gym.wrappers.NormalizeReward(env)
env = gym.wrappers.TransformReward(env, lambda reward: np.clip(reward, -10, 10))
```
However as I suggested earlier, this might be not as fast as implementing the normalization on brax's side. Another thing is directly applying these wrappers to brax environment won't work because some issues with jax's device array overriding numpy arrays in the wrappers. 

A typical example is  `gym.wrappers.RecordEpisodeStatistics`, and its `episode_returns` array will be casted to a jax array, which causes problems because jax array is not mutable.

sgillen, 2021-09-16:
Ok, I was a bit busier than I expected this week, but as promised I did start comparing the ant environments this evening. 
[Here](https://gist.github.com/sgillen/2d85c8deb9966d52ff0dfe037f9ab583) is a notebook I was using that may be useful to anyone else who wants to compare and tweak the envs.

With regards to the observations:

1. I believe all the state position and velocity information match up. For Mujoco it seems to be: z + quaternion for the torso (5), 8 joint angles, dxyz/drot (6) for torso, 8 more joint velocities, which matches exactly what brax has.  
2. The contact information is where big differences appear. Brax ignores the contact forces for some internal bodies that present in the Mujoco model, this accounts for the difference in observation size (The brax team was already aware of this). 
3. I'm not sure what the ordering for the contact forces is in brax. It doesn't match what mujoco does (see the notebook linked above) and it also doesn't seem to match up with the bodies in env.sys.body_idx.keys().
4. No matter the ordering , the magnitude of the force and and moment are substantially different, but that may be because of the difference in mass mentioned below.  

If the goal is to make as faithful representation of mujoco envs as possible (which IMO it shouldn't necessarily be) then we will at least need to address the following:
1. The mj ant starts life suspended .75m in the air, the brax ant at .5
2. mj adds a relatively large amount random noise to its initial state on reset.
3. Inertial parameters for the two envs are different. Does brax have a way to infer an inertia from geometry? This is what mj does.
4. Torque limits appear different 300 in brax vs 150 in mj (units? That would be a lot of N*m)
5. Will need to find out what brax integrator settings are closest to an rk4 with dt = .01.
6. May also need to tune friction parameters, which will probably need to be done empirically. 

TLDR: For the ant the difference in observations is in the ordering and number of contact forces. To make them match exactly we would need to re order the existing forces, and insert some dummy, zeroed elements into the observation. Should also probably adjust the mass, inertia, and torque limits.

sgillen, 2021-09-16:
@vwxyzjn good to hear about the normalization wrapper, I agree that the normalization and clipping should all be done on the brax side. This makes things awkward with respect to saving and loading environments / agents, since it will make brax a special case for gym, sb3 etc. Related, I also think that if the brax envs aren't going to be extremely fast that it would be to use pybullet.

erikfrey, 2021-09-16:
@vwxyzjn we recently started using a similar Wrapper concept for wrapping envs in Brax, inspired by Gym.  e.g. `EpisodeWrapper` collects episode statistics and sets done at the episode boundary, and so on:

https://github.com/google/brax/blob/main/brax/envs/wrappers.py#L43

I don't think it would be too hard to make the brax API mirror what gym is doing, and still keep it all on device.

erikfrey, 2021-09-16:
@sgillen this is super helpful - thanks for putting together this thorough comparison.  I hear you that our envs don't need to be exactly 1:1 to MuJoCo's - that said, we'd be happy to prioritize any fixes to the differences you brought up, according to whether they:

- impact training curves significantly (e.g. we noticed that adding noise to initial states does sometimes impact training)
- produce a more pleasing gate (e.g. perhaps re-adding the contact force reward penalty)
- ... some other reason?

Of the differences you found, do you have a suggestion for which might be the most important to address?

benelot, 2021-09-16:
I agree with @sgillen on the tasks, but would reorder to:
1. add back in the contact force reward/penalty
2. adjust the mass, inertia, and torque limit
3. add the wider distribution to initial state

On 1: If we want to copy the previous env, we need it, whether it helps with training or not, otherwise we diverge.
On 2: Is there any reason these were set in the brax ant env the way they are? Torque limit looks like the result of f(mass, inertia, mujoco_engine_details), so we should be able to set similar ones to mujoco. If they can not be adjusted exactly, I would suggest to fall back to the metric of "similar learning curve". In pybullet I once looked at the metric of "similar observation distribution shape" which says something about in which observational manifold the ant moves.
On 3: This is certainly important for higher robustness of the learned policy. Especially in the humanoids, adding some noise during testing but not training easily messes them up.

On my side I started to play a bit with brax and built some initial version of the humanoid standup but ,being on vacation, I am not done yet. I plan to begin building a first version of all required mujoco envs next week in brax just to see how they perform. Then we can do the same for every env as @sgillen did for ant.

jkterry1, 2021-09-16:
Just to confirm, does the list of inconsistencies include the list of bugs in MuJuCo that we want to make sure that we aren't reproducing that I sent?

sgillen, 2021-09-16:
@erikfrey I agree with @benelot list on what to prioritize. They will probably impact training, making the environment slightly harder if anything, but also closer to the original. The contact reward might lead to more pleasing gaits but it's hard to say.

benelot, 2021-09-17:
@jkterry1 possibly means those:
(according to Antonin Raffin)

* the broken halfcheetah is well known (blog post from 2018): https://www.alexirpan.com/2018/02/14/rl-hard.html
and was even more broken recently: https://twitter.com/natolambert/status/1369139391130607625

* for the walker, I thought I saw a comment from Erwin saying it was heavier to actually avoid having it run but I cannot find it again, I can probably find a video of the Mujoco walker running (the envs where inspired by roboschool: https://github.com/openai/roboschool)
found it!
it's in the OpenAI blog post: https://openai.com/blog/roboschool/
* according to OpenAI blog post, there was also the ant and the humanoid (but the humanoid is still quite un-realistic anyway)
* otherwise, there is the Swimmer that changed behavior (but this can be solved by using a high discount factor): https://github.com/hill-a/stable-baselines/issues/500

jkterry1, 2021-09-17:
@benelot that's the list, thanks a ton

erikfrey, 2021-09-21:
Can confirm that our HalfCheetah is at least not broken in the ways discussed in those blogs.  In fact this is something we had to address in our paper comparing our envs to Mujoco's.  See section [E1 in the appendix](https://arxiv.org/pdf/2106.13281.pdf) for a brief discussion about this problem.

That said, I am quite prepared for folks to find new and interesting bugs as these envs get more attention!  We'll be happy to address them when they come up    :-)

We are 90% done on hopper.  If someone would like to take a pass at Walker2d or Swimmer, please let me know.  Otherwise we'll get to them soon.

erikfrey, 2021-09-24:
Quick update - we now have the Hopper env, and tomorrow we will land Walker2d.  We'll also add them soon to the colab with good default hparams.  Other things in flight:

- We've added back the contact force penalty and shown it works well for Ant, confirming for others and then will push
- @erwincoumans is making great progress on a simple, small software renderer that we will use for env.render
- Wider distributions to initial state is also in progress

jkterry1, 2021-10-20:
Per the meeting, we still need the following things before merging into Gym:

Adding missing environments:
Swimmer (Benjamin Ellenberger)
Standup (Brax team)
Inverted pendulum (Daniel Freeman)
Inverted double pendulum (Daniel Freeman)

Remove 0s where applicable (Brax team)

erikfrey, 2021-10-20:
OK more updates:

- Wider distributions to initial state is in!  We jitter qpos and qvel with the exact same amounts as the current gym mujoco envs
- We now have a colab that demonstrates image rendering: https://colab.sandbox.google.com/github/google/brax/blob/main/notebooks/environments.ipynb
- image rendering is now hooked up to the gym env `render` function for `mode='rgb_array'`
- humanoid standup is in progress!

erwincoumans, 2021-10-29:
> As a sidenote, rendering images does significantly slow down the throughput.

Yes, the CPU pytinyrenderer is not intended to use during training, only afterwards to see the rollout.

2 things to make it much faster:
half the resolution (width and height), and disable anti-aliasing
```
Image(image.render(env.sys, [s.qp for s in rollout], width=160, height=120, ssaa=1))
```
@erikfrey  What is needed to make those efficiency changes when using a Gym environment wrapper?

vwxyzjn, 2021-10-29:
> Yes, the CPU pytinyrenderer is not intended to use during training, only afterwards to see the rollout.

@erwincoumans That makes sense! 

I think the desired solution also depends on if rendering the whole episode is faster. If the episode has 100 frames. Is it more expensive to (1) render the image at each frame, or (2) render the images for all the 100 frames as a batch?  

`gym.wrappers.RecordVideo` takes approach (1), which takes an `rgb_array` from brax at every step. So if there are 1000 sub environments running, approach (1) could considerably slow things down since all the other 999 envs are basically blocked. 

Maybe approach (2) is more preferable because you can have a wrapper that caches the states for the episode, when that episode ends, the wrapper spits out a video or an HTML. This way the other 999 sub environments are not blocked at all *during the rollouts*.  #

vwxyzjn, 2021-10-29:
Here is a quick example.

Imagine we have 10 envs, 10 steps (unrolls), and record 1 frame takes 1 second, and each step takes 1 second. 

Then approach (1) would take `num_envs * (t_record + t_step) * num_steps = 10 * (1+1) * 10) = 120s`

Whereas approach (2) would take `num_envs * t_step * num_steps + t_record * num_steps = 10 * 1 * 10  + 1 * 10 = 110s`.

jkterry1, 2021-10-30:
Because I forgot to post it earlier, here's my list of remaining needed features in Brax per the last group call we had:

-Swimmer
-Remove unnecessary inheritance regarding hopper
-Joint dislocation
--Adjust joint stiffness
--Generalized constraint system
-Proof of concept with fast training in PyTorch
-Brax compatible wrappers

cdfreeman-google, 2021-11-03:
Just as a little mini-update on joint dislocation: Currently, Brax handles joint constraints using springy forces, but you can also rephrase these kinds of equality constraints as velocity-level updates which directly update the velocities of a system to satisfy constraints (instead of relying on strong springs to enforce the constraints).  There's a bit of a duality to these approaches, but it turns out it's much easier to reduce jitter in the joints with velocity-level methods.  As a proof of concept, here's Ant running around with its joints being governed by this new method:
[image]

erikfrey, 2021-11-08:
OK!  I have uploaded an example of PPO in pure torch, running at 150,000 steps/sec on a Colab GPU.  It can train Ant in a few minutes.  We are far from Torch experts - I suspect there is an easy doubling in performance for someone who knows PyTorch better than we do:

[Brax Training with PyTorch on GPU](https://colab.sandbox.google.com/github/google/brax/blob/main/notebooks/training_torch.ipynb)

I think is a good enough demonstration of using Brax from PyTorch via Gym, but if anyone out there would like to speed it up further, please have a go.  @vwxyzjn - your CleanRL was a helpful reference, thank you!  We tried a few things in that Colab (including using `torch.jit`) that may help bridge the perf gap in CleanRL.

jkterry1, 2021-11-11:
For those following this thread, the remaining items are:
-Swimmer (Benjamin is working on this)
-Generalized constraint system
-Brax compatible wrappers (this is waiting on me)
-PR to Gym
-Phys2D environments to replace Box2D (future work)

benelot, 2021-12-11:
We implemented a first version that is going to be in the repo soon.

benelot, 2021-12-11:
[image]

Looks pretty decent!

jkterry1, 2021-12-11:
Awesome!

jkterry1, 2021-12-11:
The remaining action items from when we last met are:

-Grid line missing in rendering problem (Erik Frey)
-See if shadow jigging can be fixed (Erik Frey)
-Generalized constraint system
-Jumpy release
-PR to Gym
-Phys2D environments to replace Box2D
-Tensorflow conversion snippet so we can create a gym wrapper for it

erikfrey, 2021-12-13:
- OK Swimmer is live in 4f1e15a6594c4f0e797071abbb32c9c9bb3fc755
- Grid lines fixed in cb59468d95c664f3fd2b3a6b92536e5cd3d5918b
- Texture jiggling I was able to repro in Ubuntu Cinnamon / Chrome, but not a problem on other platforms.  Poking around in three.js issues the consensus seemed to be "it's a driver issue" but if I find a better answer, I'll be sure to address it.
- Jumpy release underway!

jkterry1, 2021-12-13:
That's awesome!

A minor update to the Box2D environments on my end:

-PRs are up for the needed refactoring of lunar lander and bipedal walker
-People are ostensibly currently working on the needed pygame rendering rewrite
-It turns out that the car racing environment has been used in a bunch of important work that I didn't know of by David Ha and others, so that would need to be ported instead of deleted like I'd planned

jkterry1, 2021-12-13:
There's now a PR fixing the (large) outstanding bug in car racing though

jkterry1, 2021-12-21:
Updated notes before I forget:

-Generalized constraint system
-Separate Jumpy repo
-Documentation files for the new website
-PR to Gym
-Phys2D environments to replace Box2D
-Tensorflow conversion snippet so we can create a gym wrapper for it

jkterry1, 2022-02-18:
Hey, I had a discussion with a few people. The current to-do AFAIK is:

-Generalized constraint system
-Separate Jumpy repo
-Documentation files for the new website
-PR to Gym
-Phys2D environments to replace Box2D
-Tensorflow conversion snippet so we can create a gym wrapper for it
-https://github.com/openai/gym/issues/2456#issuecomment-999653317
-Supporting the non-vector API
-Upstreaming environments
-Pusher environment (Benjamin is working on this)
-https://github.com/google/brax/issues/164

jkterry1, 2022-02-25:
Plumbing fixes:
-Final merge of generalized constraint system
-Pusher environment (Benjamin is working on this)
-PR to Jumpy repo
-Fix default camera view for reacher (briefly check the others)

-PR to Gym
--Documentation files for the new website
--Adapt MuJoCo md files (https://github.com/Farama-Foundation/gym-docs/tree/main/docs/pages/environments/mujoco)
--https://github.com/google/brax/issues/164

Future:
-Phys2D environments to replace Box2D

erikfrey, 2022-02-25:
OK, we have a fix for the default camera view in 748229c02813dd151acecc4c0a1c1741b503be55.  Next time we do a version bump, it'll land in everyone's viewers.

erikfrey, 2022-02-25:
PR to Jumpy repo is here: https://github.com/Farama-Foundation/Jumpy/pull/1

bycn, 2022-05-02:
Apologies if this is the wrong place to comment, feel free to redirect.

 Presumably porting of environments hasn't been started yet, but what's the status of 2d simulation with Brax ("Phys2d")? How would physics compare to box2d physics?