google/brax #52: spring behaviour of joints

flobotics, 2021-09-05:
hi,
i build a crawler ( https://github.com/flobotics/brax/blob/custom_envs/brax/envs/crawler5.py ) but when i train it and then visualize the trajectory, the joints went "forward-backward" like a spring ?

Here is a gif ( https://github.com/flobotics/brax/blob/custom_envs/crawler5-1.gif )  where you see how it is moving.

I tried different settings in joint description (spring_damping,etc)  but nothing changed this behaviour ?

What can i do ?   Also it seemed that the joints do not rotate the -60/+60 degrees as they could ?

thanks

flobotics, 2021-09-11:
i tried with version 0.0.5, now it looks more "artistic" :)   but the spring-behaviour is still there, i dont know how to get rid off it.  Also the physics are strange, or?

Here is a gif   https://github.com/flobotics/custom_ros2_documentation/blob/main/images/google-brax/crawler5-2.gif   of this code  https://github.com/flobotics/brax/blob/custom_envs_0.0.5/brax/envs/crawler5.py

erikfrey, 2021-09-11:
That is a very cool creature you have built      :-)

re: "strange" physics, the moment/axis of rotation might seem a bit off, as the large box is flipping around when the small arm actuates.  To make this movement look more realistic, you might want to try lowering the mass of the small arms compared to the large box (I noticed they are all 1.0 right now)

Also keep in mind that the system ignores collisions between parent and child joints, so where you are seeing some interpenetration right now, it is probably better to limit the joint angles than to rely on primitives.  I'm not seeing any issues with the joint angles in that gif, but if you think there is a problem, please do open a separate issue with a reproable use case.

re: springy joints, yes this is on our roadmap to address.  We are looking into techniques such as soft constraints: https://box2d.org/files/ErinCatto_SoftConstraints_GDC2011.pdf

In the mean time you can try increasing the stiffness of the joint, but if you raise it too high the system will explode.  This can be mitigated by playing with damping/num physics substeps.

flobotics, 2021-09-11:
hmm, thats frustrating.  I set the masses to the real values, e.g. a servo weights 54grams, so i set mass: 0.054 and so on.

But now with this code ( https://github.com/flobotics/brax/blob/custom_envs_0.0.5/brax/envs/simtoreal_three_servo_arm_ax12a.py )  it only produces a grey-image after training ? 

Also the training does not show anything in the chart output, so i think it is going NaN ?

I tried again alot of different settings, remembers me to https://github.com/google/brax/issues/44

Also i need to_tf_model which does not work, i tried it again here https://github.com/google/brax/issues/25

What do i wrong ? The creature wants to go on a walk :)

[image]

thanks for your help

EelcoHoogendoorn, 2021-09-12:
The nans are indeed an example of the system 'exploding' mentioned above. The current numerical integrator is not unconditionally stable; that is, with the wrong parameters, this kind of thing is going to happen. It will require tuning to find stable and otherwise acceptable values.

flobotics, 2021-09-12:
i made it "working" , but looks strange.  If my creature could do this, it would be great :)

The stiffness of the first joint seemed "strong" , but of the next they move, which they should not.

The other joints rotate more than they should -90/+90. You can see it at the end.

Here is a gif https://github.com/flobotics/custom_ros2_documentation/blob/main/images/google-brax/crawler5-3.gif
 and a video todo start,stop,etc. to see it better than the gif.

It would be nice to have the https://github.com/google/brax/issues/45  ,as if you need to play with dt and substeps, the training time increases. If you decrease num_timesteps and episode_length, you got a extreme short video which does not show you whats going on.

erikfrey, 2022-03-21:
Closing this issue, but feel free to reach out with any new problems.

closed 2022-03-21 by erikfrey