google/brax #53: Passive rigid objects?

namheegordonkim, 2021-09-09:
Hi,

I noticed that the config parser does not like any of the bodies in the protobuf environment definition missing joints or actuators. However, I would like to spawn physically-based spheres in the environment that can be interacted with the character but does not have joints of their own. Is this currently supported?

What I tried: I created a ball object with this protobuf string:

```
bodies {
  name: "$ BallBody"
  colliders {
    sphere {
      radius: 1
    }
  }
  inertia { x: 1.0 y: 1.0 z: 1.0 }
  mass: 10
}

bodies {
  name: "Aux 1"
  colliders {
    sphere {
      radius: 1
    }
  }
  inertia { x: 1.0 y: 1.0 z: 1.0 }
  mass: 0.001
}

bodies {
  name: "Ground"
  colliders {
    plane {}
  }
  inertia { x: 1.0 y: 1.0 z: 1.0 }
  mass: 1
  frozen { all: true }
}

joints {
  name: "$ BallBody_Aux 1"
  parent_offset { x: 0 y: 0 }
  child_offset { x: 0 y: 0 }
  parent: "$ BallBody"
  child: "Aux 1"
  rotation { x: 0 y: 0 z: 0 }
  angle_limit { min: -10 max: 10 }
}

actuators {
  name: "$ BallBody_Aux 1"
  joint: "$ BallBody_Aux 1"
  strength: 300.0
  torque {}
}

friction: 0.6
gravity { z: -9.8 }
angular_damping: -0.05
baumgarte_erp: 0.1

collide_include {
  first: "$ BallBody"
  second: "Ground"
}

dt: 0.0167
substeps: 10
```

It does spawn a ball, albeit with a weird initial position ([0, 0, 1], and then falls down). However, the joint is limited between -10 and 10, which doesn't make it a truly free object.

Any thoughts would be appreciated. Thanks!

erikfrey, 2021-09-09:
Hi there!  There should be no problems with free objects in the scene.  See the grasp environment for an example of a free ball that the hand manipulates:

https://github.com/google/brax/blob/main/brax/envs/grasp.py

If you're seeing a problem with objects that have no joints, please let us know.

namheegordonkim, 2021-09-09:
@erikfrey thanks for the quick reply. Okay, I think I was mistaken. I'll close this issue.

closed 2021-09-09 by namheegordonkim