google/brax #55: Friction per body?

namheegordonkim, 2021-09-10:
Hi,

I'm trying to create multiple bodies in the environment with differing friction coefficients. However, I couldn't find anywhere how to do this. It seems that most examples have a global friction coefficient of 0.6 but no difference across bodies.

I'd appreciate any thoughts on this. Thanks.

erikfrey, 2021-09-11:
You're quite right - we only support friction for the whole scene.  Agreed it would be nice to have!

Adding per-collider friction in the config and plumbing it through would be a pretty small PR  I've marked it as a good issue for anyone who wants to help - otherwise we'll do this in one of the upcoming releases.

namheegordonkim, 2021-09-13:
#59

erwincoumans, 2021-09-13:
Thanks for the PR! Each body (collider) may have some friction (and restitution) coefficient. For a contact, the two friction coefficients often get combined to get a 'combined friction'. You may want to make it more clear in your PR.

namheegordonkim, 2021-09-14:
@erwincoumans Aha, that makes sense.

> For a contact, the two friction coefficients often get combined to get a 'combined friction'.

Interesting. The global friction coefficient was the only thing being multiplied to impulse in `_collide()`: https://github.com/google/brax/blob/464aff88bed7c6b805a2397ef9c8f33de66be8b9/brax/physics/colliders.py#L518

So with the per-body friction/restitution scenario, I'll want to add the friction coefficients into a shared number instead? E.g. in box-plane collision, `friction = friction(box) + friction(plane)` and apply that to L518 above.

flobotics, 2021-09-21:
hi,
you probably know it already and/or it is not directly related to this issue, but at least a nice video :) 

https://www.youtube.com/watch?v=CfJ074h9K8s

PR #60 (2021-11-18): Enable per-collider friction specification
* Add optional float field "friction" to colliders; If unspecified, the global friction coefficient is applied
* Re-compile protobuf file (config_pb2.py)
* Introduce Collider dataclass to pass into _collide() and _collide_pair()
* Replace references to config.friction with per-body friction (Collider.friction)

Fixes #55

Please note:

1. I did some primitive tests within physics_test.py but observed that the simulation is currently unstable with friction due to lateral friction, no matter how large I set the number of substeps. Hence I ended up rolli [... cut]
closed 2021-11-18 by erikfrey