google/brax #56: 0.0.5 Box does not collide with sphere or capsule?

flobotics, 2021-09-11:
hi,
i got a box , capsule and sphere. I want to kick the balls (capsule or sphere) with the box, but the balls run through the boxes ?  What do i wrong ? I use version 0.0.5 

Gif is here: https://github.com/flobotics/custom_ros2_documentation/blob/main/images/google-brax/simtoreal-1.gif

code is here: https://github.com/flobotics/brax/blob/custom_envs_0.0.5/brax/envs/simtoreal_three_servo_arm_ax12a.py

erikfrey, 2021-09-11:
Hello!  Yes, we need to add box<>capsule and box<>sphere collisions.  I will create issues for both of those so we can track them on our roadmap.

erikfrey, 2021-09-11:
OK added #57, feel free to add any other thoughts there.  Thanks!

closed 2021-09-11 by erikfrey