google/brax #57: Add Box<>Sphere and Box<>Capsule colliders

erikfrey, 2021-09-11:
Probably similar to the other box colliders, where you vmap over the corners of the box and test if they are within a sphere/collider.

flobotics, 2021-09-11:
perhaps also add Box<>Box colliders ?

I added a box to my simtoreal arena and that i have not expected :)  Why is the box "forced" to/over the center ? Also the box does not collide with the servos (also boxes) ?

 Here is a gif https://github.com/flobotics/custom_ros2_documentation/blob/main/images/google-brax/simtoreal-2.gif

Here is the code to this https://github.com/flobotics/brax/blob/custom_envs_0.0.5/brax/envs/simtoreal_three_servo_arm_ax12a.py

namheegordonkim, 2021-09-13:
Well, not to go down the rabbit hole of collision, but sphere-sphere would be a good (and simple) one to add for my own use case.

FelipeMartins96, 2021-10-29:
thank you for the resources! I will take a look.
Separating spheres and capsule collidables should be avoided? I wonder if using capsule calculations for spheres create any significant overhead

FelipeMartins96, 2021-10-30:
The capsule<>box seems more complex, so i'm thinking about how to implement it, but I was able to adapt the sphere<>box implementation:

```python
def box_capsule(box: Box, cap: CapsuleEnd, qp_a: QP, qp_b: QP) -> Contact:
  """Returns contact between a box and a capsule end."""
  cap_end_world = qp_b.pos + math.rotate(cap.end, qp_b.rot)
  # transform capsule end (sphere) into local box space
  cap_rel_pos = math.inv_rotate(cap_end_world, qp_a.rot) - qp_a.pos
  cap_rel_pos = math.inv_rotate(cap_rel_pos, box.rotation)
  box_closest_point = jp.clip(cap_rel_pos, -box.halfsize, box.halfsize)
  # transform closest point back into world space
  box_closest_point = math.rotate(box_closest_point, box.rotation)
  box_closest_point_world = math.rotate(box_closest_point, qp_a.rot) + qp_a.pos

  dist = jp.safe_norm(cap_end_world - box_closest_point_world)
  normal = (cap_end_world - box_closest_point_world) / (1e-6 + dist)
  pos = cap_end_world - normal * cap.radius
  vel = qp_b.world_velocity(pos) - qp_a.world_velocity(pos)
  penetration = cap.radius - dist
  return Contact(pos, vel, normal, penetration)
  ```
  
I had to add a Box collidable and used the CapsuleEnd as a sphere, and i`m probably doing more transformations than necessary. I made some experiments and some clipping is still occurring, trying to figure out why

https://user-images.githubusercontent.com/35344539/139521688-a8e16c86-da2c-459f-adb8-8a1e6a3d8d05.mp4

https://user-images.githubusercontent.com/35344539/139521690-4496d797-030a-4e08-bebf-d838a274abfb.mp4

https://user-images.githubusercontent.com/35344539/139521696-614c3980-f7d0-4be5-9d2d-c4dda1d0ecc8.mp4

https://user-images.githubusercontent.com/35344539/139521698-00001298-676a-427f-b381-6ebd6886d7ae.mp4

flobotics, 2021-10-31:
cool, great work. Also the videos for some "proof"/visualization are well done.

erwincoumans, 2021-11-01:
@FelipeMartins96 Where is your implementation of get_sphere_penetration?

FelipeMartins96, 2021-11-03:
@erwincoumans I did not implement it, I guess I did not fully understood and made the assumption that the dist_norm was all that was needed, now that you pointed it out I see that I did not implement it correctly. I am trying implementing the capsule box collision, so I will try implementing the get_sphere_penetatrion in it.

FelipeMartins96, 2021-11-04:
I implemented the get_sphere_penetration, but I think I'm still doing something wrong, not sure about the results:
```python
def box_capsule(box: Box, cap: CapsuleEnd, qp_a: QP, qp_b: QP) -> Contact:
  """Returns contact between a box and a capsule end."""
  sphere_world_pos, _ = qp_b.to_world(cap.end)
  

  # transform sphere into local box space
  sphere_rel_pos = math.inv_rotate(sphere_world_pos - qp_a.pos, qp_a.rot) 
  # transform sphere into collidable local space
  sphere_rel_pos = math.inv_rotate(sphere_rel_pos - box.pos, box.rotation)

  box_closest_point = jp.clip(sphere_rel_pos, -box.halfsize, box.halfsize)

  dist_vec = sphere_rel_pos - box_closest_point
  distance = jp.safe_norm(dist_vec)
  normal = dist_vec / (1e-6 + distance)
  
  # START pen_norm
  # project the center of the sphere on the closest face of the box
  face_dist = box.halfsize[0] - sphere_rel_pos[0]
  min_dist = face_dist
  pen_dist = -face_dist
  pen_closest_point = jp.index_update(box_closest_point, 0, box.halfsize[0])
  pen_normal = jp.array([1., 0., 0.])

  face_dist = box.halfsize[0] + sphere_rel_pos[0]
  pen_dist = jp.where(face_dist < min_dist, -face_dist, pen_dist)
  pen_closest_point = jp.where(face_dist < min_dist, jp.index_update(box_closest_point, 0, -box.halfsize[0]), box_closest_point)
  pen_normal = jp.where(face_dist < min_dist, jp.array([-1., 0., 0.]), pen_normal)

  face_dist = box.halfsize[1] - sphere_rel_pos[1]
  pen_dist = jp.where(face_dist < min_dist, -face_dist, pen_dist)
  pen_closest_point = jp.where(face_dist < min_dist, jp.index_update(box_closest_point, 1, box.halfsize[1]), box_closest_point)
  pen_normal = jp.where(face_dist < min_dist, jp.array([0., 1., 0.]), pen_normal)

  face_dist = box.halfsize[1] + sphere_rel_pos[1]
  pen_dist = jp.where(face_dist < min_dist, -face_dist, pen_dist)
  pen_closest_point = jp.where(face_dist < min_dist, jp.index_update(box_closest_point, 1, -box.halfsize[1]), box_closest_point)
  pen_normal = jp.where(face_dist < min_dist, jp.array([0., -1., 0.]), pen_normal)

  face_dist = box.halfsize[2] - sphere_rel_pos[2]
  pen_dist = jp.where(face_dist < min_dist, -face_dist, pen_dist)
  pen_closest_point = jp.where(face_dist < min_dist, jp.index_update(box_closest_point, 2, box.halfsize[2]), box_closest_point)
  pen_normal = jp.where(face_dist < min_dist, jp.array([0., 0., 1.]), pen_normal)

  face_dist = box.halfsize[2] + sphere_rel_pos[2]
  pen_dist = jp.where(face_dist < min_dist, -face_dist, pen_dist)
  pen_closest_point = jp.where(face_dist < min_dist, jp.index_update(box_closest_point, 2, -box.halfsize[2]), box_closest_point)
  pen_normal = jp.where(face_dist < min_dist, jp.array([0., 0., -1.]), pen_normal)
  # END pen_norm

  dist2 = jp.dot(dist_vec,dist_vec)
  distance = jp.where(dist2 == 0., pen_dist, distance)
  normal = jp.where(dist2 == 0., pen_normal, normal)
  box_closest_point = jp.where(dist2 == 0., pen_closest_point, box_closest_point)

  # transform closest point back into world space
  box_point_world = math.rotate(box_closest_point, box.rotation) + box.pos
  pos, _ = qp_a.to_world(box_point_world)
  normal_world = math.rotate(normal, box.rotation)
  normal_world = math.rotate(normal, qp_a.rot)

  normal = normal_world / (1e-6 + jp.safe_norm(normal_world))
  penetration = cap.radius - distance
  vel = qp_b.world_velocity(pos) - qp_a.world_velocity(pos)
  return Contact(pos, vel, normal_world, penetration)
  ```
  
https://user-images.githubusercontent.com/35344539/140241217-e7cddf24-35d1-4e63-8b0a-7f028f2f2b02.mp4

For capsules I was thinking about creating a BoxFace collidable and base the colliders on [this capsule<>triangle collider](https://wickedengine.net/2020/04/26/capsule-collision-detection/) for each face, but not sure if it would work, specially for the sphere_penetration conditions... I'm having more trouble than I thought I would to implement this, sorry

erikfrey, 2021-11-04:
@FelipeMartins96 this is an awesome start, thank you!  Don't sweat that it's troublesome/tricky - the truth is that no one has ever written a collision function, only copy/pasted them from somewhere else.     :-)

Just quickly glancing at your code, the only warning flag is `distance = jp.where(dist2 == 0., pen_dist, distance)`, you probably instead want to check that it's less than epsilon.

We're working on some gym integration features at the moment, but after that's done we'll have time to jump in and better help you on this.  We'd love to get this feature in.

erikfrey, 2021-11-24:
@FelipeMartins96 we've added support for sphere<>box and capsule<>box in 34a70fa16497fd81ac2940f9e4a7c96bbfb31e86 - please take a look!  Thanks for your help in getting this started!

FelipeMartins96, 2021-11-26:
That`s awesome!, I wish I could have contributed more, I will take a look at that commit as I'll probably learn a lot from it. 

I will try contributing on other tasks when I have some available time.

esculapa, 2021-12-01:
Hi, do I understand correctly that box <> box colliders are to have not yet been added? It would be nice if there would be a chance for that

erikfrey, 2022-03-21:
Think this is all set - closing this issue.

closed 2022-03-21 by erikfrey