google/brax #58: Unstable / weird ball rolling and colliding behaviour

namheegordonkim, 2021-09-13:
Hi,

For context, I'm working on making a billiard simulator with Brax. I'll need good rolling / collision simulation for that. However, I'm observing strange behaviours when it comes to having two capsules roll on the ground (plane) and running into each other. See this video for example:

https://www.dropbox.com/s/zhppu5sksben1ex/Screencast%202021-09-13%2014%3A06%3A51.mp4?dl=0

The rolling ball seems to be spinning in place sometimes, and the collision doesn't seem to transfer the momentum from the first ball to the other. Is this due to something not being implemented yet, or do I have a wrong configuration somewhere? This is my config: https://pastebin.com/QPhVD8vS

Thanks.

namheegordonkim, 2021-09-13:
Just to clarify: the ball objects are capsules, not spheres. Not sure if implementing sphere-sphere collision would make this more stable, since capsules are probably not meant to be rolled around like I'm doing.

EelcoHoogendoorn, 2021-09-13:
The behavior seems physically plausible to me, if you imagine a crazy high friction coefficients, and a high mass ratio between the balls. Checking your script the ratios there appear to be 1:1 though... if so I do indeed imagine a problem with the capsules. Adding a sphere-sphere collider should be pretty easy; im currently in the process of trying to add a torus-plane collider for wheeled simulations; some stuff to figure out still but the code is quite conducive to such additions.

erikfrey, 2021-09-13:
Thanks for raising this issue.  This is a great test case that might possibly even be worthy of turning into a unit test.  I agree that the impact should transfer more momentum to the second ball!  We'll have a look at this and let you know.

erwincoumans, 2021-09-13:
There are two parts to this question:

1) the default restitution/bouncyness could be 0, in that case momentum is not fully transferred. Can you try setting it to 1 for both spheres? Can you test the collision between two spheres in space, without gravity and without plane? Using only linear velocity for one sphere, to keep the experiment simple.

2) the sphere vs ground friction could have an issue, perhaps due to wrong inertia tensor?

cdfreeman-google, 2021-09-13:
Tinkering with this a little bit, I've found some better parameters: https://pastebin.com/mtCSLvrp

Changed a handful of things:
1. Friction was WAY too high--billiards tables often have a coefficient of friction below .1 for between balls and the table
2. Fixed the inertia of the balls to accurately reflect their mass/size
3. Fixed the elasticity to be "perfectly elastic"

namheegordonkim, 2021-09-14:
Thanks all. Looks like @erwincoumans and @cdfreeman-google had the right idea of where the config was wrong.

closed 2021-09-14 by namheegordonkim