google/brax #61: Applying external forces

namheegordonkim, 2021-09-14:
Hi,

I think Brax currently doesn't support applying external force/torque to the bodies. I'm getting around this by setting linear and angular velocities with `qp.replace()`. It would be nice to have an explicit function like PyBullet's `applyExternalForce`. Any plan to implement something like this?

cdfreeman-google, 2021-09-15:
Yes, we do plan to implement this, but haven't had a good concrete need for it yet, and it somewhat complicates the `step` semantics if not done carefully.  I'll follow up here when we get to this (hopefully soon!)

namheegordonkim, 2021-10-19:
Bumping this request. External forces proved very useful for trajectory optimization in my recent project, and I'd love to use Brax for further development of this. See: https://dl.acm.org/doi/abs/10.1145/3480144

erwincoumans, 2021-11-01:
Great, closing the issue. If the external force application doesn't work for you, please file a new issue.

closed 2021-11-01 by erwincoumans