google/brax #63: Best way to calculate qp from joint angles? (Forward Kinematics)

jyf588, 2021-09-22:
Hello! Thumbs up for this interesting project. Question: what is the best way in Brax to calculate and set maximal coordinate state (qp) from generalized coordinates (joint angles)? Any pointers or rough ideas would be much appreciated.

Thanks!

erikfrey, 2021-09-24:
Hi there!  default_qp() in system.py converts generalized coordinates to maximal coordinates.  But it’s meant for joint angles specified in a config rather than provided programmatically.

I could see how having a more accessible function just for converting joint angles to qp would be useful - so I’ll leave this issue open as an improvement we could make.

In the meantime, check out default_qp()

cdfreeman-google, 2021-09-25:
You also might enjoy this issue which includes a proof of concept implementation of a shooting method for solving for torques that achieve a desired angle: https://github.com/google/brax/issues/18

As for setting the qp's directly, it shouldn't be too hard.  You'd just need to be careful to follow the definitions of the different joint angles and axes.

jyf588, 2021-09-25:
Hello, thanks for your replies! Before you get the time to make the improvement, what would be the sketchy approach in your mind if I am going to write on my own some transform function from joint angles to qps, say with JAX?

erikfrey, 2021-09-25:
You can follow the example here:

https://github.com/google/brax/blob/main/brax/physics/system.py#L99

Translated:

Traverse the kinematic tree in depth-order, for each body:
- grab the axes/angles of the joint for which the body is a child
- determine the local rotation and position of the child
- record the world rotation/position by offsetting from parent

erikfrey, 2022-01-07:
Closing, but please reopen/reply if this is still an issue for you.

closed 2022-01-07 by erikfrey