google/brax #64: Training notebook not working due to JAX updates

peilundai, 2021-09-24:
JAX pushed a new updates yesterday after which the Brax training script doesn't work anymore with the following message. I am not sure if the Brax side could help resolve the issue or we need help from JAX team. 

[image]

erikfrey, 2021-09-24:
Hello!  Thanks for pointing this out.  JAX pushed a fix in 0.2.21.  This version is not yet bundled by default in colab, so you still need to run `pip update -U jax` at the top of the notebook for it to work.  I've updated the training notebook with this temporary fix - please let me know if that works for you.

peilundai, 2021-09-27:
Thank you. It works now following the JAX update.

closed 2021-09-27 by peilundai