google/brax #74: Question about JS development

EelcoHoogendoorn, 2021-10-17:
This is probably a really silly question, but I have no experience whatsoever with JS and anything web related.

Im trying to add a torus primitive to the set of colliders. I made a simple mostly empty env to try it out in, and im using the HTML render functionality showcased in other notebook examples to try and debug my progress. Ive added colliders, changed the protobuf definition and compiled it, etc. Not much of a clue what im doing yet, but all parts of the brax repo that mention spheres now have a torus equivalent. The plane I added in my env shows up nicely; but my torus does not show up. Could be many reasons of course, but ive found at least one. Deleting createPlane in system.js in my repo, also does not stop my plane render code from working.

Further digging reveals that that html.py has a line       `import {Viewer} from 'https://cdn.jsdelivr.net/gh/google/brax@v0.0.6/js/viewer.js';`

Unless im missing something, thats not referring to viewer.js inside my repo; so it makes sense im not seeing changes made there reflected in my notebook.

Now im not sure if this is just some WIP development code that got merged, or if im missing something fundamental about the zen-of-webdev here; but if you were me and looking to make a change that would allow me to do local development on this JS, and also would stand a chance of getting merged, what would you do? Do I need to locally host my viewer.js and link the HTML to that? But then how would I merge a non-broken PR if its supposed to refer to this CDN of a past brax release?

In general, is there something im missing about doing local development? Is debugging using the 3JS viz in a notebook the way to go?

erikfrey, 2021-10-18:
Yes, please see https://github.com/google/brax/issues/33 for an example of serving the JS via a webserver.

erikfrey, 2021-10-18:
OK!  Yes it should work - please let me know if you have trouble reproducing the results in that colab.  If so I'll dust it off and take a look.

closed 2021-10-19 by EelcoHoogendoorn