google/brax #100: HTML render inverts Y axis

peabody124, 2021-11-04:
It appears there is an inconsistency in the coordinate conventions between Brax and Three.js. system.js partly handles this by swapping the Y-Z axes, but appears to leave the Y axis reversed. This can easily be seen with humanoid by raising one of the legs and comparing the outputs.

[image]

I took a crack at modifying `system.js` but can't quite get things consistent. I'm pretty sure Y should be reversed here https://github.com/google/brax/blob/main/js/system.js#L146 but couldn't get the rotations consistent.

justinberi, 2022-01-19:
I can confirm this. Using the config

```
bodies {
  name: "cart"
  colliders { box { halfsize { x: 0.05 y: 0.05 z: 0.05 } } }
  frozen { position { x:1 y:1 z:1 } rotation { x:1 y:1 z:1 } }
  inertia: {x: 1 y: 1 z: 1}
  mass: 1
}
bodies {
  name: "x"
  colliders { box { halfsize { x: 0.05 y: 0.05 z: 0.05 } } }
  frozen { position { x:1 y:1 z:1 } rotation { x:1 y:1 z:1 } }
  inertia: {x: 1 y: 1 z: 1}
  mass: 1
}
bodies {
  name: "y"
  colliders { capsule { radius: 0.1 length:0.1 } }
  frozen { position { x:1 y:1 z:1 } rotation { x:1 y:1 z:1 } }
  inertia: {x: 1 y: 1 z: 1}
  mass: 1
}
bodies {
  name: "z"
  colliders { box { halfsize { x: 0.1 y: 0.05 z: 0.05 } } }
  frozen { position { x:1 y:1 z:1 } rotation { x:1 y:1 z:1 } }
  inertia: {x: 1 y: 1 z: 1}
  mass: 1
}
defaults {
  qps { name: "cart" pos { x: 0 y: 0 z: 0} }
  qps { name: "x" pos { x: 1 y: 0 z: 0} }
  qps { name: "y" pos { x: 0 y: 1 z: 0} }
  qps { name: "z" pos { x: 0 y: 0 z: 1} }
}
collide_include {}
dt: 0.001
substeps: 0
```
Gives this in html
[image]

And this using the image renderer
[image]

justinberi, 2022-01-19:
It might make debugging easier if not conversion was made to a y-up frame. This would be tilted pi/2 radians in the html gui but the numbers reported would be one-to-one with what is coming out of brax and may make debugging easier.

erikfrey, 2022-03-23:
Thanks all for pointing this out!  There was indeed a mirroring, which is now fixed in 3a197dba959901fab16333ccde7e44e8f1981cfd.  Closing, but please reach out if you still find any problems.

@justinberi yes the raw data in the html is in Brax's native z-up representation.  The one true representation    :-)   and we convert to three.js ' coordinate system in the javascript.

closed 2022-03-23 by erikfrey

peabody124, 2022-03-23:
Awesome thank you! (p.s. I haven't forgotten #99 I'm hoping to return to the project using the residual forces in the next week and will try and rebase it on the more recent changes)

erikfrey, 2022-03-24:
@peabody124 no worries!  We actually went ahead and added external torque in 3a197dba959901fab16333ccde7e44e8f1981cfd - please let us know if you have any issues using it.