google/brax #101: Confusing gradients in simple environments

sgillen, 2021-11-09:
Hello, 

I am trying to improve apg so that it works in contact rich settings (read: the existing brax gym environments). As part of that I have been making some trivial environments to take a look at how gradients through brax behave.

I came across some behavior I don't understand for a very simple case (without any contact). I have a sphere with a "thruster" attached on the z axis, gravity is turned off and the thrust is set to zero. When I take the gradient of the thrust wrt to the ball height, I see that the gradient is growing without bound.

[Here](https://colab.research.google.com/drive/1yjeaXkNR-dkqQPNT3Tv5c3vueZoRf9b8?usp=sharing) is a notebook reproducing what I am seeing (by the way, are colab notebooks the preferred way to share code snippets around here?). 

I have been unable to figure out why this is happening, it seems like the gradient is accumulating? But I am not sure why, or why it doesn't seem to in the case where I set the thrust to a non zero value.

Is this expected behavior for brax? A bug? Or am I doing something silly?

Thank you very much. 

[image]

cdfreeman-google, 2021-11-09:
Ah yeah, this is working as intended, but it's a little bit tricky what's going on.

You aren't actually calculating the gradient with respect to the ball height, you're calculating the gradient of the loss with respect to the action applied.

`loss` returns the loss at each step, so `grads = jit_grad(action, state)` returns the gradient of the loss at each step with respect to the value of `action`.  But `action` is provided at init, and is constant over the entire trajectory.  For a short number of steps, then the ball doesn't travel as far, thus the loss is shorter, but for a long number of steps, the ball travels a great deal.  Thus, changes with respect to the action effect the loss at shorter steps asymmetrically compared to longer numbers of steps.

If this still doesn't make sense, just think about the object `loss(0.0, state)` versus `loss(0.01, state)`.  Plotted here:

[image]

Or, in other words, a small change in the action amount disproportionately effects loss further along the trajectory than shorter along the trajectory.  A numerical approximation of this derivative is of course just `(loss(0.01, state) - loss(0.01, state)) / .01`, so let's plot that too along with the gradient reported by brax:

[image]

This is a little hard to see because the numerical and real gradient are on top of each other, so here's their diff:

[image]

Which is a pretty small error.

Let me know if that doesn't make any sense!!

sgillen, 2021-11-09:
Ah, yes that makes perfect sense, thanks!

closed 2021-11-09 by sgillen