google/brax #102: Regression: Scan over env steps works in 0.0.6 but breaks in 0.0.7

louiskirsch, 2021-11-10:
Using `jax.lax.scan` for the environment loop no longer works in the latest version 0.07 (or main for that matter):
```
env = envs.create_fn(env_name='halfcheetah')()
rng = jax.random.PRNGKey(23)
rngs = jax.random.split(rng, num=10)
state = env.reset(rngs[0])

def _tick(state, rng_tick):
    action = jax.random.normal(rng_tick, (env.action_size,))
    state = env.step(state, action)
    return state, state.reward
_, rewards = jax.lax.scan(_tick, state, rngs)
print(rewards)

```
Prints the rewards in 0.0.6 and throws in 0.0.7:
```
Traceback (most recent call last):
  File "envs.py", line 188, in <module>
    _, rewards = jax.lax.scan(_tick, state, rngs)
  File "envs.py", line 186, in _tick
    state = env.step(state, action)
  File "/home/louis/git/project/venv38/lib/python3.8/site-packages/brax/envs/wrappers.py", line 87, in step
    state = self.env.step(state, action)
  File "/home/louis/git/project/venv38/lib/python3.8/site-packages/brax/envs/wrappers.py", line 61, in step
    state = self.env.step(state, action)
  File "/home/louis/git/project/venv38/lib/python3.8/site-packages/brax/envs/halfcheetah.py", line 46, in step
    qp, info = self.sys.step(state.qp, action)
  File "/home/louis/git/project/venv38/lib/python3.8/site-packages/brax/physics/system.py", line 201, in step
    (qp, info), _ = jp.scan(substep, (qp, info), (), self.config.substeps)
  File "/home/louis/git/project/venv38/lib/python3.8/site-packages/brax/jumpy.py", line 98, in scan
    carry, y = f(carry, jax.tree_unflatten(xs_tree, xs_slice))
  File "/home/louis/git/project/venv38/lib/python3.8/site-packages/brax/physics/system.py", line 179, in substep
    qp = self.integrator.kinetic(qp)
  File "/home/louis/git/project/venv38/lib/python3.8/site-packages/brax/physics/integrators.py", line 61, in kinetic
    return op(qp, self.pos_mask, self.rot_mask)
  File "/home/louis/git/project/venv38/lib/python3.8/site-packages/brax/jumpy.py", line 73, in _batched
    return jax.tree_map(lambda *x: onp.stack(x), *rets)
  File "/home/louis/git/project/venv38/lib/python3.8/site-packages/brax/jumpy.py", line 73, in <lambda>
    return jax.tree_map(lambda *x: onp.stack(x), *rets)
  File "<__array_function__ internals>", line 5, in stack
  File "/home/louis/git/project/venv38/lib/python3.8/site-packages/numpy/core/shape_base.py", line 420, in stack
    arrays = [asanyarray(arr) for arr in arrays]
  File "/home/louis/git/project/venv38/lib/python3.8/site-packages/numpy/core/shape_base.py", line 420, in <listcomp>
    arrays = [asanyarray(arr) for arr in arrays]
jax._src.errors.TracerArrayConversionError: The numpy.ndarray conversion method __array__() was called on the JAX Tracer object Traced<ShapedArray(float32[3])>with<DynamicJaxprTrace(level=1/0)>
While tracing the function _tick at envs.py:184 for scan, this concrete value was not available in Python because it depends on the values of the argument 'state'.
See https://jax.readthedocs.io/en/latest/errors.html#jax.errors.TracerArrayConversionError

```

erikfrey, 2021-11-10:
Hi Louis,

v0.0.7 introduced a new helper shim - `jumpy.py` that lets Brax users step and reset envs **without JAX**, and fall back to pure numpy instead.  This can be really useful for instantly previewing Brax environments (e.g. modifying a system config and then instantly re-rendering the system on screen) without the jit delay.

The downside is it does impose some semantic restrictions on how you should call into Brax - you have to be a bit more careful to explicitly use jax and jitted constructs only, or `jumpy` constructs only.  With this in mind, there are two ways to fix your problem.

1) jit the `_tick` function so that the jax version of `env.step` gets executed instead of the numpy version.

```python

# add jax.jit -->
@jax.jit
def _tick(state, rng_tick):
    action = jax.random.normal(rng_tick, (env.action_size,))
    state = env.step(state, action)
    return state, state.reward

_, rewards = jax.lax.scan(_tick, state, rngs)
```

2) use `jp.scan` instead of `jax.lax.scan` so that you can have it either way.  If afterwards you `jax.jit` the `jp.scan`, it will turn into `jax.lax.scan`.  If instead you call it without `jax.jit`, it will execute using numpy:

```python

def _tick(state, rng_tick):
    action = jax.random.normal(rng_tick, (env.action_size,))
    state = env.step(state, action)
    return state, state.reward

# use jp.scan
_, rewards = jp.scan(_tick, state, rngs, len=10)
```

Once we have addressed #95 and brought jit times down to an acceptable level, we will remove this `jumpy` shim as it does introduce a few weird edge cases to using Brax.  Hope this helps!

d3sm0, 2021-11-10:
Does this design choice remove the possibility of explicit evaluation of a jitted function which might become quite handy for debugging purposes? For instance, how could check the intermediate computation of the step function, and it's (maybe custom) gradient, in this case?

erikfrey, 2021-11-10:
@d3sm0 if I understand your question correctly, no, not at all.  For example, this works fine:

```python
def _tick(state, rng_tick):
    action = jax.random.normal(rng_tick, (env.action_size,))
    # jit env.step ---->
    state = jax.jit(env.step)(state, action)
    print(f'debugging something: {state.qp.pos}')
    return state, state.reward

# use jp.scan
_, rewards = jp.scan(_tick, state, rngs, length=10)
```

This outputs "debugging something: ..." 10 times.

louiskirsch, 2021-11-11:
Thanks Erik for the quick response! My issue turned out to be a bit more involved compared to what I posted above as I use haiku. I had to jit the init function explicitly (in addition to the apply function) similar to your example above as `haiku.scan` always uses `jax.lax.scan` internally. I probably can't force it it to use `jp.scan`? Unfortunately, that means compilation happens twice now (slowing things down). I guess I'll have to decouple the haiku init from using the environment.

erikfrey, 2022-03-22:
We've been using jumpy for a while and haven't encountered any JIT slowdowns, so I'm closing this issue.  But please reach out if you have any other issues using jumpy.

closed 2022-03-22 by erikfrey