google/brax #105: Mesh colliders

erikfrey, 2021-11-10:
More "real" simulations often involve complex collider geometry that must be modeled accurately in the simulator.

Convex hull is probably sufficient.  Would need to support not just Mesh<>Mesh, but also Mesh<>Capsule, Mesh<>Plane, ...

erikfrey, 2021-11-19:
As of e2ab3aaa84f1d6ae1280ec79c5d77fb44e9c3777 we now have initial support for mesh colliders, but right now have only implemented mesh<>plane and mesh<>capsule.  Check it out!

Before we call this done we'd also like to add mesh<>mesh and also some example documentation for how to use it, perhaps include it into the colabs.

But for now, here's the [Utah teapot](https://en.wikipedia.org/wiki/Utah_teapot) colliding with a plan and a cylinder in Brax:

[image]

EelcoHoogendoorn, 2021-11-27:
Interesting. Ive wondered how to elegantly implement this kind of thing in brax. Never tried my hand at it, but it occurred to me that deepSDFs might be the way to go here. Given a function to find intersection points between two SDFs, and some handcrafted SDFs for simple primitives, it should be easy to just fit SDFs to arbitrary meshes using established techniques, and you should have a very nice and generic framework, that should paralelize just fantastically on the TPU.

Some googling shows [this](https://arxiv.org/pdf/2110.01614.pdf) hot off the press. Nothing remarkable here, just an implementation of the above. But the performance comparison is interesting. It seems to compare favorably in terms of compute/memory tradeoff; and I think the compute would look even better if youd consider it in the context of a parallelized simulation running many actors batched in the same env.

EelcoHoogendoorn, 2021-11-27:
Speaking to the notion of 'good enough'; seems like the current box-box collision implementation only handles box-corner contacts; and as such probably yolos the edge-edge contact scenario entirely. Would be an interesting exercise to reimplement the box-box logic as more generic convex hulls, implemented as 'handcrafted deep SDFs'; a convex hull is just the max over a bunch of linear projections, after all. That would make improving edge-edge handling easy; just spawn some extra vertices on the edges.

esculapa, 2022-01-12:
Hi, can you share a code for the teapot or some other example of using meshes? I am trying to simulate lifting a box with outer boxes, but there are no box vs box colliders.

erikfrey, 2022-03-17:
Yes certainly, please see the mesh test here:

https://github.com/google/brax/blob/main/brax/tests/physics_test.py#L305

Config specifies a mesh path which is loaded from disk, in this test's case here:

https://github.com/google/brax/blob/main/brax/tests/testdata/cylinder.stl

We currently support mesh<>plane and mesh<>capsule, as demonstrated in the tests.  Mesh<>mesh development is in progress.  Hope that helps!

taochenshh, 2022-03-17:
thanks, @erikfrey! will check it out.

btaba, 2023-02-10:
Closing this issue as we now have general convex colliders now implemented:

https://github.com/google/brax/blob/f0de0e56126b7cd7cdf941f0fc02f3480a02716a/brax/v2/geometry/contact.py#L441-L454

closed 2023-02-10 by btaba