google/brax #108: TPU/GPU visualization for learning from pixels

erikfrey, 2021-11-10:
Brax's rendering is CPU-only at the moment.  For agents with vision, it would be useful for rendering to happen on accelerator so that training can stay fast.

[Vispy](https://vispy.org/) may be useful, but a fast native jax renderer would be even better for data on device compatability.

masud99r, 2022-01-19:
Thanks for the great simulator. Any update on TPU/GPU visualization to enable learning from pixels?

yardenas, 2022-04-15:
Hi!
I would be happy to try and help, any ideas where to start?
Could [this](https://blog.evjang.com/2019/11/jaxpt.html) be of any use?

erikfrey, 2022-04-19:
Hi @yardenas and @masud99r - JAX rendering is still an area of active interest for us!  One of those libraries might be a good starting point, although you'll also need to generate meshes from our brax primitives like capsules.

You can find examples of how to do that in our current cpu-only renderer code: https://github.com/erwincoumans/tinyrenderer

If you get on-device rendering going via JAX, please share with us a colab so we can check it out!

yardenas, 2022-04-19:
Thanks @erikfrey for the hints!

So just trying to get a better understanding: basically we would need to implement functions like ```create_capsule``` ([e.g. here](https://github.com/google/brax/blob/8a3ea6065866f687cf657d4fc5c91d002a75f72f/brax/io/image.py#L69)) which takes in the ```radius```, ```half_height``` etc, but have it written in JAX, right?

SimiPixel, 2023-01-31:
> Brax's rendering is CPU-only at the moment. For agents with vision, it would be useful for rendering to happen on accelerator so that training can stay fast.
> 
> [Vispy](https://vispy.org/) may be useful, but a fast native jax renderer would be even better for data on device compatability.

I am currently looking in this direction for my personal projects. I made a post there how one might do this efficiently. Might be interesting here too. [Issue over at VisPy](https://github.com/vispy/vispy/issues/2448)

JoeyTeng, 2023-05-29:
Hi, I have created a pure JAX implementation and an adapter layer mimicking the behaviour of existing [pytinyrenderer](https://github.com/erwincoumans/tinyrenderer). The code will be released under Apache-2.0 License [here](https://github.com/JoeyTeng/jaxrenderer/tree/v0.1.0). I will be working on a PR to replace current visualisation code in Brax soon.

A working example can be found under the root directory, e.g., [this](https://github.com/JoeyTeng/jaxrenderer/blob/v0.1.0/example.py). I will implement them in Colab tomorrow.