google/brax #114: HTML Rendering Bug/Undesired Behavior in JupyterLab

sgillen, 2021-11-13:
Hello, I've been experiencing issues when making multiple calls to the html renderer while using jupyterlab (3.1.18, under both chrome and firefox).

I can produce the behavior by running the following code in my instance of jupyter lab (with each #%% representing a new cell)

```python

#%% Cell 0
from IPython.display import HTML

import brax
from brax import envs
from brax import jumpy as jp
from brax.io import html

#%% Cell 1 

env_name = "ant" 
env_fn = envs.create_fn(env_name=env_name)
env = env_fn()
state = env.reset(rng=jp.random_prngkey(seed=0))

HTML(html.render(env.sys, [state.qp])) # First call, works fine

#%% Cell 2

HTML(html.render(env.sys, [state.qp])) # Second call, creates a blank space under cell 2, and silently adds a new underneath the viewer created when running cell 1 (which is invisible without scrolling).
```

As the comment explains, every call after the first results in the new viewer being added underneath the first cell. If that's confusing I've attached a screenshot that explains what I'm seeing. Even worse, if I open a new tab/notebook and attempt to render there, it will add a third viewer to the first cell of the original tab!

This differs from the behavior seen in colab notebooks, where each call to html.render makes a viewer in the cell it's called in (which is the desired behavior here, for me). I believe the issue is conflicting div ids for the viewer windows, perhaps colabs are doing something to give unique ids for html generated in different cells? 

Either way I have a fix that works for me here: (! I will edit this in a few seconds !) which adds a uuid to the viewer name.

[image]

erikfrey, 2021-11-16:
Hi Sean,

Did you try an IFrame?

```python

from IPython.display import IFrame

...

IFrame(src=html.render(env.sys, [state.qp])) # First call, works fine
```

I see your PR and agree that making the viewer id unique fixes the immediate problem, but it may cause other problems down the line to allow use cases where multiple HTML files (with opening/closing tags for html and body, multiple script includes, etc) are appended into the same html file.

You're right that in colab this is different!  I think we are taking for granted that in Colab, outputs are automatically injected into an IFrame, I believe.

sgillen, 2021-11-18:
Hmm I can't get that to work, does that work for you?. Modifying your idea slightly, I can get something that almost works:
```python
from IPython.display import IFrame

...

IFrame(src = f"data:text/html,{html.render(env.sys, [state.qp])}",  height=480,  width=900)

```
However it looks like somewhere along the way the src string is being truncated, and so I just get a blank box.

I've also tried 

```python
HTML(f"<iframe srcdoc={html.render(env.sys, [state.qp])} title=test height=480 width=960></iframe>")
```

Which also gives a blank screen (and a warning, that I should be using the built in IFrame).

Probably one of these approaches can be made to work, but I don't have time to dig into it further right now. 
I am going to keep using the solution in the PR for now, but totally understandable if you prefer not to merge it.

erikfrey, 2022-03-22:
Closing this, but happy to revisit if we hear from more JupyterLab users that visualization is not working.

closed 2022-03-22 by erikfrey