google/brax #116: Force- or velocity-based actuators

RedTachyon, 2021-11-19:
From what I can tell, right now actuators can only work by applying a torque to a joint. Is there a sensible way of acting upon the bodies present in the scene by applying a force to them, or perhaps by directly setting their velocity instead?

erikfrey, 2021-11-19:
Hi @RedTachyon , you can add linear forces through the act array by adding a Force to the config:

https://github.com/google/brax/blob/main/brax/physics/config.proto#L148

See inverted_pendulum for an example of how to use it:

https://github.com/google/brax/blob/main/brax/envs/inverted_pendulum.py

We will also soon add external torques too, see #99

AlessandroZavoli, 2022-02-16:
This issue also relates to [Issue #163](https://github.com/google/brax/issues/163) : Aerodynamic forces/torques can be approximated as function of v squared

erikfrey, 2022-03-23:
We now support both external linear force and torque inputs, as of 3a197dba959901fab16333ccde7e44e8f1981cfd.  Please try it out!  Closing this issue, but feel free to reach out with questions/bug reports/etc.

closed 2022-03-23 by erikfrey