google/brax #117: Weird simulation interactions when the frozen floor body does not have the mass property set

benelot, 2021-11-19:
While developing the swimmer env, I realized that the mass property of the floor body is really important, even though it is frozen and thus is acting like something with infinite mass. First of all, I found the first body of the swimmer to be slightly overlapping with the floor, then later as I applied a random action to see if the swimmer works correctly, its body parts moved around randomly. Adding ```mass: 1``` resolves the issue completely.

So my question is: Does the mass property in frozen: all bodies make sense somehow and if not set, should it be set to a default value to avoid simulation issues?

erikfrey, 2021-11-19:
Great find.  I agree mass on a frozen body should not matter.  We will have a look.

cdfreeman-google, 2022-03-28:
Hey Benjamin! I don't suppose you happen to have a repro for this, do you?

benelot, 2022-10-11:
I would assume you could remove the mass property from the body in the
swimmer env colab and see if it still persists. In the pusher env, I saw
that it just throws out some division by zero mass into the console if the
property is not set. I do not have a direct repro for this, sorry.
It happened when the frozen floor body does not have the mass property set
(as generated by the mujoco converter).