google/brax #119: MuJoCo converter example colab

erikfrey, 2021-11-19:
@benelot proposed in #68 that a colab showing off the MuJoCo converter would help more users discover that Brax has this capability.  Benjamin, if you build such a colab, we will happily include it and link to it from the top level README.

The MuJoCo converter is still not turnkey - after creating a config you sometimes have to adjust parameters such as joint stiffness.  @cdfreeman-google has some rules of thumb that we could add to such a Colab as notes.

/cc @benelot @cdfreeman-google

benelot, 2021-12-11:
I will prepare something and PR it.

yardenas, 2022-04-06:
@erikfrey hi!
Are there any other things than joint stiffness that you would recommend checking/tuning after running mujoco_converter script on mujoco's .xml files?

And also thanks for developing brax, it's super cool :)