google/brax #120: URDF inertia and mass conversion

limbryan, 2021-11-19:
There is a [TODO: on the URDF converter](https://github.com/google/brax/blob/main/brax/tools/urdf.py#L217) which currently leaves out converting the mass and inertial information. I wanted to do this and push this but when working on it, I noticed that the urdf format contains more information (see example below) than the brax body config format which just has `inertia { x: 1.0 y: 1.0 z: 1.0 }, mass: 1`.  Not sure how to proceed - would be great if this feature could be completed. Thanks!

```
<inertial>
      <!-- CENTER OF MASS -->
      <origin rpy="0 0 0" xyz="0.0405 0 0"/>
      <mass value="0.1872"/>
      <!-- box inertia: 1/12*m(y^2+z^2), ... -->
      <inertia ixx="3.154e-05" ixy="0" ixz="0" iyy="1.896e-04" iyz="0" izz="1.972e-04"/>
 </inertial>
```

cdfreeman-google, 2021-11-29:
Currently, Brax only supports storing the inertia in a diagonal representation--i.e., the inertia tensor in the coordinate system where x'y''z''' point along the principal axes for that primitive.  The upshot of this is that Brax only really supports spherical-like inertias at the moment.  It's on our todo-list to expand inertial support to more complicated bodies.

As for why nothing happens when you run, you probably want to modify the `done` condition in your step function, because the body that is being checked with this line:

`done = jp.where(qp.pos[0, 2] < 0.2, x=jp.float32(1), y=jp.float32(0))`

is actually at .13 at init, so the system is auto-resetting instantly.  If you set this instead to:

`done = jp.where(qp.pos[0, 2] < 0.05, x=jp.float32(1), y=jp.float32(0))`

then everything works great.

(though I might suggest changing the angle limits to +- 180 instead of 360, and setting `gravity: { z: -9.8 }` in the config.)

limbryan, 2021-11-30:
Thanks for the clarification. That was really helpful! Do you think a good/best estimate (make-do) for the converter right now would then be to just represent the full inertia tensor in a diagonal representation by just assuming the off-diagonal terms as zero?

cdfreeman-google, 2022-03-23:
Ahh apologies for not responding here---yes, that's a good interim solution.  We have plans to represent non-spherical inertias eventually, but even then, we'd probably store the principal moments (so we only need to store a vector), and then rotate it where needed.

cdfreeman-google, 2022-03-23:
Closing this for now, feel free to reopen if you have more issues!

closed 2022-03-23 by cdfreeman-google