google/brax #121: Frozen model during ppo.train

bhuv-z, 2021-11-22:
Still figuring my way through Brac, and I'm in the process of porting a pybullet model (stored as an URDF) into a brax model. 

The following is the config currently in use: https://github.com/GeoffreyMClark/MicroRaptor/blob/main/custom_envs/bolt.py

During `ppo.train` the model remains frozen and no movement can be visualized, even though the joints seem to be configured properly (based on the existing humanoid environment) Any help in troubleshooting this would be appreciated.

Thanks

erikfrey, 2021-11-24:
Hello Bhuvan,

Previewing your system, I see this:

[image]

The default QP positions the base link at y=0.63.  But in your environment, you done an episode if base link is < 0.65:

https://github.com/GeoffreyMClark/MicroRaptor/blob/main/custom_envs/bolt.py#L65

I haven't actually tried training it, but I would expect in your training that you are seeing all truncated episodes due to this - it's probably done'ing instantly and resetting after the first step, which would probably explain the frozen state you are seeing.

You probably want to adjust those done conditions to enforce whatever behavior you're trying to achieve - in the humanoid training env, it's to stop the humanoid from falling over.

erikfrey, 2022-01-07:
Closing, but please reopen/reply if this is still an issue for you.

closed 2022-01-07 by erikfrey