google/brax #127: Bug: capsule/sphere-plane collision working unreliably

fgolemo, 2021-11-29:
I'm trying to simulate a pool billiards tabletop that I've modeled with a plane as floor and 4 planes surrounding it, facing inward.
If I now spawn a ball and bounce it off the walls, sometimes the collisions work well:

(left is top-down view, right is side view)

[image]

...but sometimes it doesn't work and the ball gets stuck in the wall:

[image]

[image]

I've made a Colab to reproduce the issue:
https://colab.research.google.com/drive/1flnseQcjarIYM4G_rECTEAaps-kXPoey?usp=sharing

If anybody has any pointers, that'd be greatly appreciated.

Best,
Flo

namheegordonkim, 2021-11-29:
Can confirm this is the case for my own billiard simulator as well.

cdfreeman-google, 2021-11-29:
Well that's strange.

I've repro'd, and also confirmed that adding a small amount of `baumgarte_erp: .01` to the config stops this from happening.  I suspect this is arising because of imprecision in how the contact is being resolved.  I also confirmed that raising elasticity a little bit removes this behavior.

I'll look into this a bit, but a full fix might not appear until we take another hard look at collision physics.

fgolemo, 2021-11-29:
@cdfreeman-google thanks for looking into this!

I've tried adding `baumgarte_erp= .01` at the end of the `bouncy_ball = brax.Config(...)` line and it works sliiightly better. I'm still getting one of the starting velocities stuck in the wall:

[image]
 
As for higher elasticity: what value did you find working?

namheegordonkim, 2021-12-01:
I played around with my environment, and this seems to get rid of the sticky behaviour:

```
friction: .1
gravity { z: -9.8 }
angular_damping: 0.00
baumgarte_erp: 0.01
elasticity: 0.9
dt: 0.0167
substeps: 240
```

I think it's most to do with the number of substeps, but it would be nice to not have to spend so much simulation time to get rid of this artifact.

namheegordonkim, 2022-02-10:
Update: my current workaround is to rely almost completely on translation rather than rotation--I think the behaviour has to do with angular momentum not changing reliably during collisions. My up-to-date configuration is:

```
friction: .1
gravity { z: -9.8 }
angular_damping: -10
baumgarte_erp: 0.01
elasticity: 0.9
dt: 0.0167
substeps: 20
```

Essentially, very slippery surface with pretty harsh angular damping. Trajectories look a lot better with this setting.

cdfreeman-google, 2022-03-28:
This should be fixed at head now with the newest collision / elasticity physics.  Feel free to reopen if you're still seeing weirdness!

closed 2022-03-28 by cdfreeman-google

namheegordonkim, 2022-05-12:
I'm afraid the issue is still there with dynamics_mode=pdb. See the GIF below.

https://giphy.com/gifs/n0RmuXGLuJnBysxKiH

(don't worry about balls suddenly moving up--it's just simulating pocketing.)

namheegordonkim, 2022-05-13:
Hi @erikfrey , thanks for the super prompt reply!

Here's a minimal colab example I cooked up:

https://colab.research.google.com/drive/1h3vfUSAO_KK3A41MkCLHFx5cGcd7D2YZ?usp=sharing

I've pushed substeps to 20, but funnily enough, setting it to something like `substeps: 2` makes the artifact go away at narrower shot angles.

cdfreeman-google, 2022-05-13:
Thanks for the quick repro! 

I've also found that reducing `solver_scale_collide:` to about `.2` in the config helps for your setup.  This looks like the velocity level update is being overly aggressive in how it's applying damping/restitution.  Will do some spelunking today.

cdfreeman-google, 2022-05-16:
Yeah so when applying per-body materials, you have to also give it an elasticity, otherwise it'll default to 0.0 (and thus hit the wall, and dissipate all energy in the normal direction).  So, ironically, low substeps only "worked" here because it made it break in exactly the right way :P

namheegordonkim, 2022-05-17:
Ok, that explains it. Shouldn't default values for per-body material properties be set by global friction/elasticity rather than 0? In fact, I thought this was the behaviour for friction.