google/brax #128: Cannot take gradient over action with humanoid step function.

chaihahaha, 2021-12-01:
```python
import jax
import brax.envs.humanoid
r = brax.envs.humanoid.Humanoid()
rng = jax.random.PRNGKey(0)
def generate_step(s1, env_step):
 return lambda a: env_step(s1, a).reward
s1 = r.reset(rng)
a1 = jax.random.normal(rng, (r.action_size,))
step = generate_step(s1, r.step)
grad_step = jax.grad(step)
print(grad_step(a1))
```
Then I get:
```
jax._src.errors.TracerArrayConversionError: The numpy.ndarray conversion method __array__() was called on the JAX Tracer object Traced<ConcreteArray([570.79987    -3.1874492 236.74734  ])>with<JVPTrace(level=2/0)>
  with primal = DeviceArray([570.79987  ,  -3.1874492, 236.74734  ], dtype=float32)
       tangent = Traced<ShapedArray(float32[3]):JaxprTrace(level=1/0)>
See https://jax.readthedocs.io/en/latest/errors.html#jax.errors.TracerArrayConversionError
```

erikfrey, 2021-12-02:
Howdy @chaihahaha - I think you are running into how brax uses jumpy.py to allow Brax to be used without JAX.  When using jax.grad, you should jit the function you want to take a gradient over so that it uses all the jax primitives under the hood.  So this should get you going:

```python
import jax
import brax.envs.humanoid

r = brax.envs.humanoid.Humanoid()
rng = jax.random.PRNGKey(0)

def generate_step(s1, env_step):
  # add a jax.jit here:
  return lambda a: jax.jit(env_step)(s1, a).reward

s1 = r.reset(rng)
a1 = jax.random.normal(rng, (r.action_size,))
step = generate_step(s1, r.step)
grad_step = jax.grad(step)
print(grad_step(a1))
```

chaihahaha, 2021-12-03:
It worked! Thank you so much for your help!

closed 2021-12-03 by chaihahaha