google/brax #132: AttributeError: module 'brax.training.ppo' has no attribute 'make_params_and_inference_fn' on colabs

benelot, 2021-12-10:
I was just training and visualizing swimmer and suddenly realized the method make_params_and_inference_fn disappeared. Then I checked the tutorial colab and realized they are broken too:
https://colab.research.google.com/github/google/brax/blob/main/notebooks/training.ipynb#scrollTo=fgB52sgjDhvi

How to fix it?

@erikfrey

benelot, 2021-12-10:
Ok, looks like it is just

```
model.save_params('/tmp/params', params)
inference_fn = ppo.make_inference_fn(environment.observation_size, environment.action_size, True)
params = model.load_params('/tmp/params')
```
according to this: https://github.com/google/brax/commit/46720002c49cfc287dc45e196227990f0be1e3f9#diff-386fba1221b2e3452fe0c5234da9660a7e985cf811964f9cd29b8f11d5332579

erikfrey, 2021-12-10:
Good catch!  Updated the colab in a1fe625eaf8080eb100d24008c5796241585b5e0

closed 2021-12-10 by erikfrey