google/brax #133: Checkerboard for floor object

benelot, 2021-12-10:
While training the swimmer I have a hard time to see if it even moves if I simulate a viable policy manually. Is there a way to enable a checkerboard floor texture to make this easier?

benelot, 2021-12-11:
As Erwin tells me, there should be a grid on the ground, but it is not always visible (visible when looking top-down, but any other angle it is not there).

Suggestion: dont z buffer it or at least somehow in front of the floor object.

erikfrey, 2021-12-13:
I was able to repro this on ubuntu chrome.  Poking around, I found a few related issues in three.js where the maintainers suggested it was some driver issue around drawing 1px width lines.

So I switched it to a checkerboard.  If your chrome is like mine, there is still a jitter that is impacting the texture and the shadow map texture - this doesn't show up on OSX chrome.  But at least now you can see movement - try it out!

erikfrey, 2022-01-07:
Closing this as we've migrated to a checkerboard, but please reopen if anything else seems broken.

closed 2022-01-07 by erikfrey