google/brax #136: Collision with frozen box not working

AlexanderKoch-Koch, 2021-12-14:
It seems like the collision between capsule and box doesn't work when the box is frozen. However, it does work when I replace the box by a frozen sphere or capsule. 
This is the result when I set ```frozen {all: True}``` for the box:
[image]
And this is the result without the frozen attribute:
[image]

Here is my system config:
```
bodies {
  name: "Ground"
  colliders {
    plane {}
  }
  inertia { x: 1.0 y: 1.0 z: 1.0 }
  mass: 1
  frozen { all: true }
}
bodies {
  name: "capsule"
  colliders {
    capsule {
      radius: 0.1
      length: 0.5
    }
  }
  inertia {
    x: 1.0
    y: 1.0
    z: 1.0
  }
  mass: 1.0
}
bodies {
  name: "box"
  colliders {
    box {
      halfsize: {
        x: 0.05
        y: 0.05 
        z: 0.05
      }
    }
  }
  frozen {all: true}
  mass: 10.0
}
angular_damping: -0.05
baumgarte_erp: 0.1
gravity {z: -9.81}
collide_include {first: "capsule" second: "Ground"}
collide_include {first: "capsule" second: "box"}
collide_include {first: "box" second: "Ground"}
dt: 0.01
substeps: 8
```

erikfrey, 2021-12-16:
Haha - YES, I can verify that that is unphysical behavior!  We did not test frozen meshes - thanks for pointing out this bug.  We'll look into it.

EelcoHoogendoorn, 2021-12-20:
Ive ran into something similar recently; without tracing the latest master... im guessing this is because the dispatch of OneWayCollider assumes that its always the second body participating in the collision that is the frozen one (which happens to work out for box_plane and capsule_plane; which are probably the only ones that have been properly tested)

closed 2024-10-04 by btaba