google/brax #139: Alpha Update in Soft-Actor-Critic

milutter, 2021-12-15:
Hey Brax Community,
I am not an expert on SAC, but the `alpha` update within the SAC implementation looks a bit weird to me. I might also be missing some mathematical identity or jax trick that resolves the misunderstanding.

In [line #498](https://github.com/google/brax/blob/4fe55ea8a2699859de66b97275fc29a5e8aecbcd/brax/training/sac.py#L498), the `alpha_params` are defined as `log_alpha`. In [line #343](https://github.com/google/brax/blob/4fe55ea8a2699859de66b97275fc29a5e8aecbcd/brax/training/sac.py#L343), the gradient of the `alpha_loss` is computed with respect to `alpha`. The resulting gradient is used in line [line 362 & 364](https://github.com/google/brax/blob/4fe55ea8a2699859de66b97275fc29a5e8aecbcd/brax/training/sac.py#L362) to update the `alpha_params` using the ADAM optimizer. If I am understanding the code correctly, it seems that the `log_alpha` parameters are updated with the gradient with respect to `alpha` and not `log_alpha`.  Or am I missing something?

If this is actually the case, it would be very easy to fix. When looking at different open-source implementations some actually implement `J_alpha = log_alpha * [Expected Entropy - Desired Entropy]` (e.g., [Mushroom RL](https://github.com/MushroomRL/mushroom-rl/blob/62ff8ab7f1ac886d415b53a805a60175f3397071/mushroom_rl/algorithms/actor_critic/deep_actor_critic/sac.py#L343) and [MBPO](https://github.com/jannerm/mbpo/blob/ac694ff9f1ebb789cc5b3f164d9d67f93ed8f129/softlearning/algorithms/sac.py#L272)) while some other implement the original Eq. 18 from the paper, i.e., `J_alpha = alpha * [Expected Entropy - Desired Entropy]`.

m-orsini, 2021-12-16:
Excellent point! We're pushing a change, it should be in later today.

Thanks a lot!

erikfrey, 2021-12-16:
This is pushed now - thanks Michael.

milutter, 2021-12-16:
Great! Thanks for the fast fix.

closed 2021-12-16 by milutter