google/brax #140: Evaluation for VectorWrapper

Cospui, 2021-12-16:
I found it is hard to understand the `eval_unroll` function in `Training in Brax with PyTorch on GPUs` Colab notebook. The environments seem to work in an auto-reset way, while the `eval_unroll` function calculates the episodes (number of `done`) and whole rewards in the trajectories. Will this take the rewards for the last several steps into consideration, which are not done yet? This may probably overestimate the evaluation performance. Or I just misunderstood it.

erikfrey, 2021-12-16:
@Cospui You are absolutely right - we took a shortcut here, and are not properly handling truncation for calculating statistics.  I'll make a point to call this out more explicitly in the colab.

The main purpose of this colab is to show that fast training wtih brax is possible using PyTorch.  Before I wrote the colab, I wasn't sure that it was!

At some point I'd love to work with one of the popular PyTorch RL packages to support fast training too - the main challenge is making sure the replay buffers and other acceleration data never leave the device.  Once I have a good example of using an off the shelf PyTorch RL package with Bax, I'll retire this colab.

erikfrey, 2022-01-07:
Closing, but please reopen/reply if this is still an issue for you.

closed 2022-01-07 by erikfrey