google/brax #141: Inverted Pendulum Observations

milutter, 2021-12-28:
Hey,
I know that the inverted pendulum is not the main interest for Brax. However, the [current observation computation](https://github.com/google/brax/blob/661364ee04f2e31017bea40e02eee6968bde5af2/brax/envs/inverted_pendulum.py#L57) seems odd compared to the OpenAi gym and DM Control suite design choices. The observations are computed using

```
(joint_angle,), (joint_vel,) = self.sys.joints[0].angle_vel(qp)
qpos = [qp.pos[0, 2:], qp.rot[0], joint_angle]
qvel = [qp.vel[0], joint_vel]
obs = jp.concatenate(qpos + qvel)
```
If I understand it correctly, this observation is weird as 
* `qp.pos[0, 2:]` is the **constant** z-height of the cart, 
* `qp.rot[0]` is the **constant** quaternion of the cart,  
* `qp.vel[0]` is the 3d velocity of the cart, which is **constant** in 2 dimensions. 

One could easily use the more common cartpole observations, which can either be computed using

```
# 4d Observations = [Cart Pos, Angle, Cart Vel, Angle Vel]
obs = jnp.concatenate([qp.pos[0, :1], joint_angle,  qp.vel[0, :1], joint_vel])

# 5d Observations with sine/cosine features instead of the angle to avoid the +/- pi jump
obs = jnp.concatenate([qp.pos[0, :1], jnp.cos(joint_angle), jnp.sin(joint_angle), qp.vel[0, :1], joint_vel])
```

I am happy to contribute a pull request. However as it is only one line, it might be easier for you to sneak it into the next release.

cdfreeman-google, 2022-01-06:
Well look at that!  I'm fine with either convention.  It looks like the env here doesn't have the sine/cosine features: https://github.com/openai/gym/blob/master/gym/envs/mujoco/inverted_pendulum.py, so I'll probably go with that for the sake of compatibility.  Thanks for spotting this!

cdfreeman-google, 2022-01-07:
(This will be fixed in the next sync!)

cdfreeman-google, 2022-01-07:
And this fix is now live!

erikfrey, 2022-01-07:
Closing, but please reopen/reply if this is still an issue for you.

closed 2022-01-07 by erikfrey